winder/Universal-G-Code-Sender #924: Windows 10 lastest Java UGS Wont start

nickm324, 2018-02-28:
When run the start-windwos bat starts to initialize and then get this error and it exits.

 Feb 27, 2018 7:30:06 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Log location: C:\Users\rippe\.ugs\UniversalGcodeSender.json
Feb 27, 2018 7:30:06 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Loading settings.
Initializing...
Exception in thread "main" java.util.MissingResourceException: Can't find bundle for base name resources.MessagesBundle, locale zh_CN
        at java.util.ResourceBundle.throwMissingResourceException(Unknown Source)
        at java.util.ResourceBundle.getBundleImpl(Unknown Source)
        at java.util.ResourceBundle.getBundle(Unknown Source)
        at com.willwinder.universalgcodesender.i18n.Localization.initialize(Localization.java:68)
        at com.willwinder.universalgcodesender.i18n.Localization.initialize(Localization.java:55)
        at com.willwinder.universalgcodesender.MainWindow.<init>(MainWindow.java:117)
        at com.willwinder.universalgcodesender.MainWindow.main(MainWindow.java:244)

winder, 2018-03-01:
Thanks for the report, this should now be fixed in the latest nightly build. Thanks @breiler

closed 2018-03-01 by winder

nickm324, 2018-03-01:
Thank you for the quick turn around.

winder, 2018-03-01:
@nickm324 Maybe you can answer this question. UGS currently has localization for Chinese (simplified) but the locale ID is **zh_CHS**. Do you know if it would be correct to change **zh_CHS** to **zh_CN**?

winder, 2018-03-01:
@nickm324 thanks, just renamed the file. Hopefully the next zh_CN user will have a much better experience. 636ba8e6d65365fd5ad168fc35b9eb03d748de8c

nickm324, 2018-03-01:
I bet they will. :)