winder/Universal-G-Code-Sender #932: Issues with probe plate

jahnj0584, 2018-03-04:
Sometimes it runs just fine. Here is my macro
G91 G54 G38.2 Z-50 F200; G0 Z1;G38.2 Z-1 F50; G92 Z14.93; G53 G0 Z0; G54

If it alarms, it says "probe fail. Not in expected state before starting probe cycle etc"

Just wondering what it could be? The plate has one red wire attached from A5. The ground is coming from the frame of the machine (continuous)

I found it also alarms out if I touch the plate, but it may be a coincidence? It was already in the motion of probing.

winder, 2018-03-04:
Looks like you're seeing [error code 4](https://github.com/gnea/grbl/blob/master/doc/csv/alarm_codes_en_US.csv):  `Probe fail. Probe is not in the expected initial state before starting probe cycle when G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.`

From the GRBL code and your description, it sounds like this is happening because the probe pin is already. I'm not sure why. I'd try adding some dwell commands before the probe operations and see if that helps.

jahnj0584, 2018-03-04:
Adding G4 P4 does nothing to help.

jahnj0584, 2018-03-05:
Fixed it. Bare shield in the cable was touching the plate when it was laying down. I figured it out after it'd probe fine if I held the plate in the air....

closed 2018-03-05 by jahnj0584