winder/Universal-G-Code-Sender #936: Major bug with G38 results

jahnj0584, 2018-03-07:
Very dangerous! If you cancel a job before it reaches the M30, UGS seems to reset the Z axis value to the offset or whatever it was before you probed (I assume?)

I use these tiny 0.5mm bits, with plastic collars that are repeatable to 0.02mm. I can just swap bits without rezero'ing and carry on EXCEPT if I probe these for Z instead of just using the paper trick. I just swapped one, and had the spindle dive down 10mm INTO my work piece. Choice words were spoken. 

This is definitely a bug one how UGS is reacting to the offsets applied via a G38 macro.

jahnj0584, 2018-03-07:
did a test without spindle on. -1.3mm Z became a -18.25 if I canceled the job......

jahnj0584, 2018-03-07:
https://youtu.be/WpiIB9ml-og

winder, 2018-03-09:
Duplicate of #860 

I think whats happening here is GRBL resets to `G54`, it causes some very unexpected behavior. UGS should set the machine back to the previous coordinate system to avoid this.

closed 2018-03-09 by winder

jahnj0584, 2018-03-15:
https://github.com/gnea/grbl/issues/397#issuecomment-373209213