winder/Universal-G-Code-Sender #937: What pendant can be used?

adigitaldj1, 2018-03-07:
I would like to use a pendant, what would be a good one to use?

winder, 2018-03-09:
UGS doesn't have any special pendant handling at this time.

closed 2018-03-09 by winder

petmakris, 2018-03-20:
I have a HB04 XHC pendant in hand, wireless, which outputs via HID. If UGS supports plugins or a way add functionality I could write this interface.

Came here looking for plugin support in UGS for this matter

winder, 2018-03-20:
@petmakris UGS has plugin support, you can setup the boilerplate with this maven archetype:

https://github.com/winder/Universal-G-Code-Sender/tree/master/new-module-archetype

petmakris, 2018-03-20:
Thanks winder, I already found this out.

jasonhamilton, 2020-02-19:
Curious if you made any progress on this.  I have a HB04 pendant that I would like to use.