winder/Universal-G-Code-Sender #938: Request: Visualizer Origin Select

CNCQuilter, 2018-03-07:
I am using UGSPlatform along with GRBL and hand-derived gcode to control a consumer level quilting machine. The gcode for sewing continuous designs/curves is derived either by hand or by Visual Basic and/or EXCEL mathematical functions. For me, it would be advantageous to be able to change the 0.0, 0.0 origin in the Visualizer Window. Out of a long ago acquired habit (and serendipitous for quilting), my origin starts at the upper left hand corner with X+ to the right and Y+ from top to bottom.
Is it possible to reset the 0.0. , 0.0 origin in the Visualizer to the top left corner?

Thank you.

winder, 2018-03-09:
The visualizer isn't really intended to represent the machine at this point, it is purely centered around the gcode. Being able to define your machine space would be a cool feature at some point, and would make setup operations more interesting in the visualizer.

Also inverting the Y axis isn't something that I can support right now.

To have the visualizer show up this way you would need to configure your gcode in the X+ and Y- space.

closed 2018-03-09 by winder

CNCQuilter, 2018-03-10:
Thanks for your reply, I really appreciate your efforts.

Dave Kender

jahnj0584, 2018-03-10:
Gcode ripper can swap your axis or even apply a rotation to the file

proofpoint.com/v2/url?u=https-3A__github.com_winder_
Universal-2DG-2DCode-2DSender_issues_938-23issuecomment-
2D371859074&d=DwMFaQ&c=3buyMx9JlH1z22L_G5pM28wz_Ru6WjhVHwo-vpeS0Gk&r=
zO1BysYTt4UFmSFEv9o0SbJim_z9Tb-G4UxWKuM9N7I&m=R2tOXSnPnuZMJ-jyzk4P1Wsr9zh9-
OY_WKWSYxfn5D4&s=2Kv8B1Y_BEjh650IpuCtaFZySC9g62Pycz5Mjyu1tKs&e=>, or mute
the thread<https://urldefense.proofpoint.com/v2/url?u=https-
3A__github.com_notifications_unsubscribe-2Dauth_AScC7lC0CCBP70WBp-
5FWDlUHjNkbl4J72ks5tcqpzgaJpZM4Sgh65&d=DwMFaQ&c=3buyMx9JlH1z22L_G5pM28wz_
Ru6WjhVHwo-vpeS0Gk&r=zO1BysYTt4UFmSFEv9o0SbJim_z9Tb-G4UxWKuM9N7I&m=
R2tOXSnPnuZMJ-jyzk4P1Wsr9zh9-OY_WKWSYxfn5D4&s=egvKe3kCnbY4JYHm-
fhPfKYV5GNwCOGQdUZ17AFLX7k&e=>.