winder/Universal-G-Code-Sender #939: Jog controller design

breiler, 2018-03-09:
With some help from @dastultz, we've started building a new jog controller plugin.

The intention is to make it more scalable than the previous component and have a consistent look with the DRO. It's still a bit of a work in progress but I would like your comments and feedback.

The normal view:
[image]

The bit more compact view:
[image]

vMeph, 2018-03-09:
Looks awsome, if you keep button press will it keep moving until you release button also?

winder, 2018-03-09:
Looks great!

This next comment is a bit outside the scope of this redesign, but I'm curious if you saw this. There were some ideas in #779 for a way to make jogging more responsive to keyboard commands (i.e. how it works in the classic GUI). My original idea was to have a button that makes this panel pop up in a modal way, but maybe there is another solution?

The problem is that keyboard commands are sometimes swallowed (I think by the visualizer, and the text editor). When this happens keyboard shortcuts like jogging don't always work. Also simple single-character commands don't work well with the NetBeans keymappings.

breiler, 2018-03-10:
@vMeph I first implemented it like described here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging#joystick-implementation
It would jog as long as you pressed the jog-button, on release a jog-cancel command was sent. It worked pretty well but had the side effect that it would be very hard jogging to an exact position. So I scrapped that, it's now working as before. I was thinking of adding a button-pressed-event where continuous jogging is activated after the button  is being pressed for ~500ms.

@winder yes I saw it, this could be a good next step. I saw that sourcerabbit have button for giving the panel focus: https://raw.githubusercontent.com/nsiatras/sourcerabbit-gcode-sender/master/Images/ControlForm.png

jahnj0584, 2018-03-10:
> joystick-implementation
> gcode-sender/master/Images/ControlForm.png

Is this the new colour scheme UGS is heading towards? Could we have options
for the standard netbean tan + black? I love black and cyan but i also want
my control to be as simple as possible.

breiler, 2018-03-10:
That's a good input. It would actually make the coding a lot more simple if we used standard buttons and we would use more of the panel for the buttons (less borders). 

Just for comparison, this is what it would look like in standard colors (it still needs some adjustments):
[image]

**You guys decide:**
:-1: for plain colors like the current jog panel
:+1: for new color theme alá DRO

vMeph, 2018-03-10:
All looks good would be great if there was a way to have prebuild themes, i personaly like back themes, the prebuild themes thst UGS have all looks basikly the same

dastultz, 2018-03-10:
I am of course biased towards the new color scheme. I use it in combination with the Darcula theme. I do make color choices based on the current theme, so I suppose we should have the new colors in a custom look-and-feel. I didn't want to force people to use a dark theme to get the new colors. My goal was a consistent look to the DRO that works on light and dark themes.

Generally I've been trying to make UGS look better. People will perceive the application as a whole as being better if the UI looks good and we should get more users. The plain colors look very boring to me.

vMeph, 2018-03-10:
Dracula theme is cool, but it seems it has some conflit with the splash logo, it will cut the pic of splash logo

breiler, 2018-03-10:
For all you vampires out there, we could easily add support for darker themes using the plain layout. With almost no code change (except for giving the labels a foreground color) and adding icons with _dark-suffix we can support both default and custom dark themes:

[image]

I'm all for both. On the one hand crisp and consistent layout such in @dastultz DRO, and on the other less code to maintain and more standard Java/Netbeans components.

jahnj0584, 2018-03-10:
For a controller, the priority in order should be

Function: limited by grbl so that's  "done" barre some wizards and plug
ins.

Stability: still an issue, especially with M30's and cancelled jobs.

Aesthetic: controllers should be bare as it's all about the locations and
visualizer. It doesn't have to be flashy. Machine DROs are just black
numbers on a white screen. Haas controllers are also just blue text on an
off white background. I'm aware that adding colours doesn't make it
unstable but it does make it harder to find numbers when your reaction time
changes a saved part to a scrapped one.

Im happy as long as both are an option. Who knows, i may end up using the
blue one but would rather not be strong armed into it.

jahnj0584, 2018-03-10:
Is left right the best way to go about setting speeds and step sizes? What
if we want to jog 1mm then 650? Do we hold the dial down for 649 increments?

dastultz, 2018-03-10:
@vMeph I am not seeing any difference in behavior of the splash screen for Darcula and Mac OSX LAFs. I don't have the knowledge to help you with that issue.

The main questions on the table at this point should not be the colors and specific graphic design elements but the layout and readability. I don't think creative colors are necessarily in conflict with readability. The scheme I have developed is "high contrast" which is most important. Color blind people still need to be able to tell what's what. I don't think we need to fall back to a simple black and white in order to be readable.

I think @jahnj0584 raises a good question about the UI controls for step/feed. For parity, I suppose the new design should have the numbers editable.

The big thing I want to hear about is the units button. It takes up a lot of space and doesn't really fit in well. What is the argument for running the machine/job in inches, and then jogging in mm? If the jog panel is in MM and I move 10mm to the right, the DRO may be showing the position in inches because that is the mode GRBL is in.

I think the Jog panel should respect the units the machine is in. Who disagrees and why?

winder, 2018-03-11:
@dastultz 
Personally, I like that the unit's are visible on the jog panel and I wouldn't want them to use the current g20/g21 setting. For various reasons the files I use aren't consistently imperial or metric, so when moving the machine after a job I'm not always sure what mode is even active.

I think the fact that UGS exposes all these basic things makes it really easy for new users to jump in without knowing about g20/g21.

@breiler 
I would be happy with either option you've laid out. Given the choice between both, I'd probably pick the simpler code that looks decent in both light and dark themes.

I'm also really interested in the continuous mode you implemented. Maybe if the units button could be replaced with a menu popup that allowed choosing units and enabling/disabling continuous mode?

jahnj0584, 2018-03-11:
I like that last option. Even if you design in inches or metric (as i often
do) your CAM should be static with the units it exports

carneeki, 2018-03-14:
Why not keep the jog panel inputs as unitless?

You move over 1, the controller takes it to mean 1 of whatever the native
unit is on GRBL? (eg 1mm, 1in, 1 smoot, whatever).

breiler, 2018-03-15:
Ok, how about this.

I figured that using separate step settings is for more experienced users so it will not be enabled by default. It's still available in the preferences and in the step size menu. This will make the button a bit smaller by default. Couldn't we use the machine reporting units (from the firmware setting) as the jog unit?
[image]

I've also added labels for the buttons so that it can't be mistaken which axis the button control. It now supports continuous jogging if the button is pressed for more than 500ms. It behaves as before if it's single clicked.
[image]

Dark theme is also supported:
[image]

This jog panel isn't enabled by default, it must be selected using the window-menu:
[image]

jahnj0584, 2018-03-15:
Love it Joacim. When can i get it?

dastultz, 2018-03-15:
I approve! Good work on this, thanks.

vMeph, 2018-03-18:
@breiler is there any way to adapt the keyboards arrow keys to work with the jog movements, it seems ugs dont let you add them has choise, the left right it allwed but up and down dont let you

jahnj0584, 2018-03-18:
Meph it works if you add a trigger. I use CTRL so you never hit it by
accident

vMeph, 2018-03-18:
[image]
doesnt let you

jahnj0584, 2018-03-18:
Maybe it was shift i don't remember. It's second nature to me

vMeph, 2018-03-18:
wont let you shift either....

breiler, 2018-03-18:
It could work to let the window have it's own keyboard listeners and only listen for the arrows. But then we would need something gain focus in the panel. I don't think that will be intuitive. =/

I got the same warning as you when trying to assign new shortcuts. But it works if I try set CMD+DOWN  (in my case CTRL+Up and CRTL-Down is used by the operating system) and then click somewhere else in the window.

[image]

winder, 2018-03-20:
Using a meta key was my workaround to get the shortcuts to work reliably with Netbeans Platform.

closed 2018-03-20 by winder

ct2djk, 2019-07-21:
Olá colegas.
ando a procura de uma aplicação que me de a conversao da dobra de tubos para coordenadas  XYC, para poder inserir na maquina que dobra tubos, mas tem software propriatario atraves de um automato.

um abraço

Hello colleagues.
and looking for an application that allows me to convert data to XYC coordinates, to be able to insert the machine that doubles the tubes, but has proprietary software through an automato.

a hug