winder/Universal-G-Code-Sender #940: JSSC + Java 9 + Windows

winder, 2018-03-09:
There seems to be a driver issue with JSSC + Java 9 + Windows
Defaults here: https://github.com/scream3r/java-simple-serial-connector/issues/116

UGS could use the patched version of `jssc.jar` in the LaserCut project which may resolve the issue in some cases:
https://github.com/wholder/LaserCut/blob/master/lib/jssc.jar

breiler, 2018-04-27:
Tried replacing JSSC with [jSerialComm](http://fazecast.github.io/jSerialComm/) because the project seems more active.
It seems to work when using the classic version. However I cannot get the nbm packaging to work... 
As a side note, it lists serial ports with the prefix `cu` which could fix the issue #1016.

@winder do you have any ideas why this isn't working?
https://github.com/breiler/Universal-G-Code-Sender/commit/6e1915bb85fe03a2f7caec46b36474fd258da0ea

I get the build error: 
> [ERROR] Failed to execute goal org.codehaus.mojo:nbm-maven-plugin:4.1:manifest (default-manifest) on project ugs-platform-ugslib: Execution default-manifest of goal org.codehaus.mojo:nbm-maven-plugin:4.1:manifest failed. IllegalArgumentException -> [Help 1]
> org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.codehaus.mojo:nbm-maven-plugin:4.1:manifest (default-manifest) on project ugs-platform-ugslib: Execution default-manifest of goal org.codehaus.mojo:nbm-maven-plugin:4.1:manifest failed.

breiler, 2018-05-29:
Does anyone run Windows and Java 9 that can try out the new connection driver library. 

Change from JSSC to JSerialComm in the settings:
[image]

And remember to refresh the list of ports:
[image]

breiler, 2018-07-31:
I was able to get a hold of a Windows machine.  Unfortunately it was running 32-bit Windows 7, and I wasn't able to reproduce the problem. However it worked with both JSSC and JSerialComm. I'm wondering if we could do an experiment setting JSerialComm as a default for new users?

winder, 2018-07-31:
I'd be fine disabling JSSC completely and switching over to JSerialComm for everybody.

closed 2019-03-04 by breiler

kmorosky50, 2020-12-27:
Hello, using UGS classic. Getting same error. "No serial ports found"

Using WIndows 8. Machine is pulled in via USB port. But unable to detect in UGS. 

I have GIT, JAVA installed. 

ANy suggestions? 

Thank you!

lulomro, 2021-11-22:
For anyone who can help with the problem of not seeing COM ports.
It worked with WIN XP, java 8.0 and ugs 2.0.8.
Solution for me=> 
Options = driverconexion = change from JSerialCom to JSSC

PR #1198 (2019-03-04): Bugfix/connection driver
Fixes #940 and makes JSerialComm as the default driver. This will only have effect if the user is missing the ugs-settings directory and the default settings are loaded.