winder/Universal-G-Code-Sender #942: Platform not working on RPi3

kmsorensen, 2018-03-13:
I'm trying to get Platform to work on my RPi3.  I'm running java version 1.8.0_65 and the latest GRBL and UGS.  Classic UGS is working.  Platform worked once initially and crashed.  Now a mostly white screen comes up with 2 visualizer drop downs and some of the top menu bar.  Some of the top menu bar drop downs "work", but nothing physical happens (servos silent) and the screen remains white.  When started in terminal the last line of code is /resources/firmware_config/smoothieboard.json.  I found some suggestions on FAQ-pi3d 2.22 regarding finding and linking files, and followed them with no improvement.  It's perhaps obvious, but I just barely know what I'm doing here.  I'm thrilled to get Classic to work.  Would be nice to get Platform to work too.  Any suggestions?  Thanks!

kmsorensen, 2018-03-13:
Additional info:  Before I linked the files the last line of code when trying to start Platform was libEGL warning: GLX/DRI2 failed to authenticate.  After linking the files as suggested in FAQ pi3d the last line of code changed to  /resources .... etc.

breiler, 2018-03-13:
I've downloaded the latest nightly build (Mar 11, 2018) and I have no problems starting the platform version except for the Visualizer which is an known issue (#47).

Could you upload the log file for UGS. 
This should create a file in your home directory with the last lines from the log: 
`tail -n1000 ~/.ugsplatform/2.0-SNAPSHOT/dev/var/log/messages.log > ~/messages.log`

kmsorensen, 2018-03-13:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/1808963/messages.log)

kmsorensen, 2018-03-16:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/1817672/messages.log)

kmsorensen, 2018-03-16:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/1817676/messages.log)

closed 2018-03-16 by kmsorensen

cdsmith66, 2018-04-03:
I see this is closed, but I'm curious if it was remedied as I'm having a similar issue. When running platform in terminal I've also had it end with /resources/firmware_config/smoothieboard.json, but when I ran it a second time the last line of code was /resources/firmware_config/test_2.json.

carneeki, 2018-04-04:
The messages you describe are normal informational messages. I suspect you
might have a separate / different issue.

I run on an RPi 3. I pull the source and get updates quite frequently, are
you running the latest UGS code? If so, you probably want to open a new
issue.

The other thing to check is if UGS is actually running, after the last
line, are you given a prompt again to type more commands? If not it's
possible UGS is running off-screen somewhere).

cdsmith66, 2018-04-04:
@carneeki I found a few already submitted issues that were similar so I didn't think it warranted a new issue, but I may just have to open a new one. 

I am running the latest nightly build and when running UGS platform in terminal it launches the splash screen, the terminal shows that it is loading a bunch of services and then it display the code I listed above. It doesn't prompt me to enter any additional commands and the splash screens stops at "Done loading modules".

It seems to be similar to issue #918 which was remedied, but it seems to persist on my system . Sorry for my lack of knowledge as i'm new to this , but do I need to do any additional steps on the Pi? I've confirmed that the Pi is running Java 8 as well. 

Thanks for the help.

carneeki, 2018-04-05:
There might be something further up in the messages, which is logged to `messages.log` inside `~/.ugsplatform/2.0-SNAPSHOT/dev/var/log`. There might be something in there that points to the problem. If it is too much, you can attach the log file here and I'll look over it and see if I can find it.

Out of curiosity, which vendor for Java are you using? Official from Oracle? OpenJDK? (Pasting the results of `java -version`). I've had success with OpenJDK 7 and Oracle JDK8.

cdsmith66, 2018-04-05:
@carneeki I looked through the message file and didn't really see anything obvious, but I also don't know what I'm looking for, so I have attached the file for you to review. 

Here is the information on Java and someone else installed Java, so I'm not sure which vendor was used. 

`pi@raspberrypi:~ $ java -version
java version "1.8.0_65"
Java(TM) SE Runtime Environment (build 1.8.0_65-b17)
Java HotSpot(TM) Client VM (build 25.65-b01, mixed mode)`

Thanks again for the help and let me know if you want me to throw this in another ticket. 
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/1880955/messages.log)

cdsmith66, 2018-04-05:
@carneeki well, you were spot on, I moved those files and the latest nightly build had no problem launching. I'll have to keep that in mind for future updates.

Thank you very much for spending the time to help me through this. I've learned a lot about navigating around a Linux system in the last 24 hrs.

winder, 2018-04-05:
Thanks @carneek!