winder/Universal-G-Code-Sender #943: will not start on OS X 10.13.3

davidsi, 2018-03-13:
install from july 28 stable build UGS platform, running java 9.  Get the following error when trying to run. 

~/Desktop/ugsplatform/bin >./ugsplatform
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/Users/david/Desktop/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

then shuts down

breiler, 2018-03-14:
I've reproduced your problem and got this error in the log:
`tail -n100  ~/Library/Application\ Support/ugsplatform/dev/var/log/messages.log`

```
-------------------------------------------------------------------------------
java.lang.SecurityException: setContextClassLoader
	at java.base/jdk.internal.misc.InnocuousThread.setContextClassLoader(InnocuousThread.java:104)
	at org.netbeans.ModuleManager.updateContextClassLoaders(ModuleManager.java:476)
	at org.netbeans.ModuleManager.<init>(ModuleManager.java:169)
	at org.netbeans.core.startup.ModuleSystem.<init>(ModuleSystem.java:109)
	at org.netbeans.core.startup.Main.getModuleSystem(Main.java:169)
	at org.netbeans.core.startup.Main.getModuleSystem(Main.java:150)
	at org.netbeans.core.startup.Main.start(Main.java:307)
	at org.netbeans.core.startup.TopThreadGroup.run(TopThreadGroup.java:123)
	at java.base/java.lang.Thread.run(Thread.java:844)
```

It wouldn't help setting the `-J--illegal-access=warn` in the `etc/ugsplatform.conf`. I think your best options are either to install Java 8 or use the nightly build which uses NetBeans 8.2 instead of 8.1.

davidsi, 2018-03-14:
yeah, that boots better.  Thx

breiler, 2018-03-20:
May we close this issue?

closed 2018-03-20 by winder