winder/Universal-G-Code-Sender #944: Soft Limit Alarm Problems While Sending G-Code

SRFirefox, 2018-03-14:
OS: Mac OSX El Capitan and Ubuntu 16.04.2
UGS Platform Nightly 201609300101
GRBL 1.1f

When a streamed g-code command throws a soft limit alarm (alarm:2) during a g-code stream, the sender properly stops the stream and displays a window showing the error. After closing the window and doing a soft-reset on GRBL, UGS displays the "Grbl 1.1f['$' for help]" prompt and then takes no more inputs from UGS except for a soft-reset; the alarm state is unclearable. The duration counter in the send progress also continues to count up. To be able to continue to use UGS do anything with the controller, the Arduino needs a hard reset and even then some functions like the jog controller don't work until closing and opening UGS again. I'd expect to be able to clear the alarm and start working again rather than having to hard reset, re-home the machine, and then start using it again.

Basic steps to re-produce the problem:
1. Boot the Arduino and connect with UGS
2. Home GRBL controlled machine
3. Load g-code file into UGS and move the machine where it will hit the soft-limits when running the g-code.
4. Wait until program tries to move through the soft limits.

When I have more access to the machine I'm working on (a day or two) I can do some more characterization and better logging of the issue.

mrsamdk, 2018-03-17:
I can confirm experiencing the exact same behaviour. Everytime i've encountered a soft limit alarm and done a soft reset, then jog controls are dimmed out/not clickable regardless of what I done afterwards. Only quitting/re-starting UGS will make jog controls active again.

breiler, 2018-03-20:
Thanks for an excellent bug report! The proposed fix may resolve this issue.

breiler, 2018-03-26:
@SRFirefox, @mrsamdk we have made some changes together with #759 that could resolve this issue as well. If you have the time I would really appreciate if you tried out the latest nightly build and verified that the issue is gone.

SRFirefox, 2018-03-27:
Can do. I should have some time in the next couple days and I will report back with my findings.

BallscrewBob, 2018-04-07:
Not sure if this is 100% linked.
Similar behaviour in 2.0 nightly Jan 21st.
Just went to latest JAVA
Win 7 PRO x64, AMD x4 bulldozer, 16GB ram. 

Most of the same symptoms but tend to see it more if I STOP a running file. 
File itself does not matter but larger ones often provide a better trigger point and make it easier to see the bug. 

Its the fact that the JOG options are disabled. That means a total restart of UGS and the faint hope I can do a return to zero.

Soft reset , unlock etc will not help in these events either.

mrsamdk, 2018-04-08:
@breiler I tested the latest nightly build and after doing a soft reset, the jog controls are now available again (before I needed to reboot UGS entirely for it to become available).

I only manage to test triggering a soft limit alarm and then clicking the soft reset button. I'm not sure if other symptons was mentioned in @SRFirefox 's description

breiler, 2018-04-11:
I'm no longer able to reproduce the error so I'm closing the issue.

closed 2018-04-11 by breiler