winder/Universal-G-Code-Sender #945: Command Line Documentation

HasanEdain, 2018-03-14:
I am interested in running G-Code-Sender from the command line. I have looked at the developer documentation, and see only the line:  (from: http://winder.github.io/ugs_website/dev/getting_started/) "The UGS Classic and Platform interfaces can also be run from the command line. These commands should be run from the root directory."

Is there a place to look for the command set / command parameters? Just a little hint in the documentation could point me in the right direction, thanks.

breiler, 2018-03-14:
That section in the documentation is meant for starting the applications when developing.

There's only one parameter in the UGS Classic for opening files: `--open | -o <filename>`
(and of course setting Java-parameters such as memory settings)

The UGS Platform is based on NetBeans which has a bunch of parameters but not much to do for controlling machine:
http://wiki.netbeans.org/FaqStartupParameters

So it depends on what you want to do. 😄

carneeki, 2018-03-14:
Possibly a separate issue, and it possibly is already resolved, but if one were to specify a filename in argv[1], it would be cool for to open a file, or if .ngc files (and similar) could be associated for opening with UGS. That last one gets a bit messy though as we probably don't want to start a second instance, but maybe put the file into a queue (such as if we are running multiple toolpaths we can have the next file queued for manual tool changes).

winder, 2018-03-15:
@carneeki take a look at the "Workflow Helper" plugin, it sort of does what you're describing.

@HasanEdain currently there is no CLI mode for UGS

closed 2018-03-15 by winder

HasanEdain, 2018-03-16:
I was looking (hoping) for a more complete scriptable control, I may have to remember my Java and see if I can wire such a thing up. What I would like to do is to make a small web server that I can connect to with my iPhone (I am an iOS developer), and send a g-code file to. Have the web server USB connected to my Laser Cutter, and engrave pictures from my photo library onto stuff.

I am aware that there are other ways to do this job, but I thought it would have some hack value :)

I will next look at some of the UI code in the project and see how much work it would be to separate that from the UI and drive it from a command line.

Thanks for the help. It is good to know that if I go down a harder road, at least I checked the easy first :)