winder/Universal-G-Code-Sender #946: Gcode window for read and/or editing & Job's time estimation ?

Myke974, 2018-03-14:
Hey there !
First, developpers&others, Thank you for the awesome job and Thank You for sharing opensource !
I'm actually using Universak Gcode Platform with cheap cnc shield V3 
[image]
and so far I got to use my homemade cheap CNC in plotter mode
[image]
and since few days in Laser mode ! ! Everything is working like a charm ! 👍 
WooooooooooooooT ! ! 🤣 

Second : 2 questions : 
1] I can't find where to see the actual gcode of an imported file 😅 

2] Is there a way to have a job's time estimation ?

Regards

jahnj0584, 2018-03-15:
You don't see the code. UGS only streams it to the arduino due to size issues. A built in text editor is in the works 

Send progress shows what you need.

winder, 2018-03-15:
There used to be a way to export the processed gcode (i.e. the code after it has had comments/whitespace removed, decimals truncated, etc), but that feature never made it to the platform version of UGS.

@Myke974 is that the feature you're asking about with your first question?

closed 2018-03-20 by winder