winder/Universal-G-Code-Sender #950: FEATURE REQUEST: Support Logitech G13 as Pendant

RecceDG, 2018-03-15:
So this thing: https://www.logitechg.com/en-sg/product/g13-advanced-gameboard is just begging to be used as a potential low-cost, feature-rich pendant controller.

It can be configured through Logitech's keyboard interceptor to feed UGS useable keystrokes, but that doesn't support the LCD screen. Native support would be better.

There is a java wrapper for the Logitech SDK on GITHUB.

I'll be happy to help test if someone wants to code this.

Myke974, 2018-03-15:
> gameboard is just begging to be used as a potential low-cost,

My two-cents :
Low-cost ?
For me low cost is more something like this :
https://www.aliexpress.com/item/Mach3-6-Axis-USB-Joystick-Rocker-Handheld-Electronic-Handwheel-for-CNC-Carving-Machine/32810829359.html?spm=a2g0s.13010208.99999999.343.PyIocL
----
This said, I don't want to devalue your proposition in any way ;)
At contrary, I'm thanksfull ;)

winder, 2018-03-20:
Something like this is definitely possible, but I don't have any plans to work on it.

closed 2018-03-20 by winder