winder/Universal-G-Code-Sender #951: New Jog Controller Error 15 when holding down mouse button (Build 966)

kenfritz, 2018-03-16:
Today I tried using the new Jog Controller but got an "Error 15: Jog Target Exceeds Machine Travel" when clicking on any of the jog arrows and holding down the left mouse button more than 1 second. If I single click the mouse button I don't get an error. This error is not a single error but starts a chain reaction of the error repeating hundreds of time and I must Kill UGS process to get out of this problem.

I attached a screen shot showing the error.
Build 966 (UGS Plateform)
[image]

breiler, 2018-03-16:
Argh, sorry about that!

I haven't tested it with soft limits, and now that I see the error I feel stupid! 😓 

When the button is pressed and hold it will continuously queue jog commands with a quite big step length so that the machine is kept busy with processing commands. According to the GRBL documentation these steps should be kept small, but I got in to trouble when I tried that because it ended up streaming at some point which lead to that the jog cancel didn't work (maybe because the send buffer was filled).

I'll try to fix this by checking the send queue before creating a new jog command, making sure just one or two commands are queued. Then decrease the step length, maybe base it on the current units and feed rate.

breiler, 2018-03-17:
Fixed some of the problems:
* Shortened the jog size so that the soft limit isn't triggered directly
* It will now cancel the jog command if there's an error (instead of leaving the jog-loop to continuing) 

One problem still remains. There is no way to find out how many jog commands that are queued which means  there is no way of limiting the number of commands that are being sent.

breiler, 2018-03-18:
@kenfritz this should be available in the nightly build. If you get a chance to test it please let me know if it is working better for you.

kenfritz, 2018-03-19:
I have been testing today Bld# 970 jogging around and it looks fixed. I jogged long distances holding the mouse button and got no errors. I loved the diagonal arrows the best. I really like the changes you are making to the jog control feature. Great job.

closed 2018-03-20 by winder