winder/Universal-G-Code-Sender #952: Safety door issue on univ gcode platform?

tarmon01, 2018-03-16:
I’m using a safety door switch on
my DIY laser etcher.. which is supposed to shut things off when the door is open (does) and restart things back going when door is closed.. which it doesn’t...

The only thing that I can do to clear the error after the door alarm is activated (and then deactivated) is to close the serial port itself.

is that behavior expected?

Thanks,

   -TomH-

winder, 2018-03-18:
UGS doesn't have much special handling for door switches. You should see the door pin enabled in the DRO, I'm not sure what else happens on the GRBL side though.

closed 2018-03-18 by winder

tarmon01, 2018-03-18:
Hey thanks!

It is supposed to work similar to “feed hold” and “cycle start” except it uses a single input pin.

Ie when it is active it pauses the job  and when inactive it lets it go...

The pin is toggling ( i see it as a message below where it shows real time stuff ) but the big bold “ state”  line above seems to be active and not change once triggered... . is the “state” line directly monitoring the grbl state or is “state” something implemented in ug platform?  If it’s just showing “grbl  “state” I will go debug grbl..

Thanks,

-TomH-

tarmon01, 2018-03-19:
Will,

Okay some more investigation... grbl docs say the following:

Door: (New in v0.9i) This compile-option causes Grbl to feed hold, shut-down the spindle and coolant, and wait until the door switch has been closed and the user has issued a cycle start. Useful for OEM that need safety doors.
——
So when UGP sees the  “Door” state  when “?” Is issued ( prints on serial command “<Door:1”) it should continue to check the state until it sees “<Door:0” at which point UGP should issue the “~” ie. the “cycle start”command to continue.

I tested this out on a serial port and it seems to be the correct behavior.

You agree?

-TomH-

dastultz, 2018-03-19:
I have a special switch hooked up to GRBL's door pin. When I engage the switch, the cycle pauses, the state says "door", the alarm flag "door" lights. When I disengage the switch, nothing happens. Maybe after a few seconds the alarm flag clears. Then after pressing the "play/start" button on UGS twice, the cycle beings again. I'm using GRBL 1.1.

tarmon01, 2018-03-19:
So if I hit the “ play” button after a door is active and inactive it works! ... as cycle start is issued again and it continues where it stoped. This is much better than the serial port thing. 

So no changes are needed and sorry for taking up your time.

Thanks,

-TomH-

winder, 2018-03-19:
@dastultz, @tarmon01, thanks for the updates!