winder/Universal-G-Code-Sender #958: inconsisten overrides

jahnj0584, 2018-03-18:
Sometimes the overrides stay after a job is sent, and is applied to the next file sent (issue when its over 100%). Sometimes its reset and no longer active (issue when its under 100%).

Ive also had it once be applied, but the feed rate was what was programmed (no its not near my max speeds)

It should be a constant application, going off of my experience with Haas machines.

zeevy, 2018-03-19:
Grbl firmware will reset the overrides after the program end command ie M02.
If your gocode file don't have "M02" as the last line, then Grbl will not reset the overrides.

winder, 2018-03-19:
thanks @zeevy

closed 2018-03-19 by winder

jahnj0584, 2018-03-19:
All my files are with the same post and end with m30

zeevy, 2018-03-23:
m30 will also resets the feed overrides