winder/Universal-G-Code-Sender #962: 2.0 Nightly builds/ Tiny-G

booyaka716, 2018-03-20:
Is there any way the developers of this G-code sender can do some work with the Tiny-G platform? I can only move my X, Y, Z manually! Reset zero, soft reset, unlock, check mode, return to zero, home machine, and get state isn't supported yet for some reason, just like the older versions! I'd love to use the software, but you seem to like to put Tiny-G as an option, but still not able to do anything!

breiler, 2018-07-13:
I've started working on this and there should now have better support. It's still early in the development and I would really appreciate anyone testing this.

breiler, 2018-07-25:
Closing this, please open a new issue if you find any problems.

closed 2018-07-25 by breiler