winder/Universal-G-Code-Sender #966: UGS Platform 2.0 64 not opening after Windows Creators Update PLZ HELP

landbrad, 2018-03-23:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

- [ ] UGS Platform 2.0 - Nightly Build
- [ ] UGS Platform 2.0 - Stable
- [ ] UGS Classic 2.0 - Nightly Build
- [ ] UGS Classic 2.0 - Stable
- [ ] UGS Classic 1.0.9
- [ ] UGS Classic 1.0.8
- [ ] UGS Classic 1.0.7

#### Operating system

- [ ] Windows
- [ ] Linux
- [ ] Linux on RaspberryPI
- [ ] Mac OSX
- [ ] Other

#### Platform

- [ ] GRBL 1.0 or later
- [ ] GRBL 0.9
- [ ] GRBL 0.8
- [ ] TinyG
- [ ] SmoothieBoard

#### Other

breiler, 2018-03-24:
What version of Java are you using?
(Open command terminal and run `java -version`)

breiler, 2018-05-11:
Closing because additional information is missing.

closed 2018-05-11 by breiler