winder/Universal-G-Code-Sender #970: Cannot resume after GRBL error

flounderscore, 2018-03-26:
## Problem description 
With the latest code (github as of yesterday), I cannot resume when grbl returns an error caused by tool change commands.

### Expected Behavior
In the past the gcode was interrupted when the grbl board reported an error when it received a tool change command ("T10 M6"). I then swapped the tool manual manually and hit the "play" button to resume the gcode.

### Actual Behavior
Does not resume, unfortunately.

### Specifications
#### Version
Github (3/25)

#### Operating system
Linux Ubuntu

#### Platform
GRBL 1.1

breiler, 2018-03-26:
I can't reproduce your issue.

I've tried both 66ad033d65e439d18fe135f2d3cfc7f7f61d0dc5 and 515a773fa19c952acea3a69cfe0eb1c863c77f8b from 3/25 and the current master 7138f8ff2f8f658e1b16249586c3bbe4023c04c2 from today with both single stepping and streaming.

I got an error as you described, and after I hit "run" and the program continues.

I was running this gcode, using GRBL 1.1f on a Mac.
```gcode
G17 G20 G90 G94 G54 M0 M5 M9
G0 Z0.25 F5
X-0.5 Y0.
G1 Z0.
(arcs...)
G2 X0. Y-0.5 I0.5 J0.
(tool change)
T10 M6
(tool changed)
G2 X-0.5 Y0. I0. J0.5
```

What is your exact GRBL version and which commit have you compiled?

flounderscore, 2018-03-26:
Hi Joacim,
thanks for looking into the issue. I'll try again tonight to make sure I
didn't miss anything.

flounderscore, 2018-03-27:
I managed to reproduce the issue yesterday. The console displays " Cancelling file transfer." and "Finished sending file in 00:00:00".
After pulling the latest code from Github the issue disappeared.

Closing the issue.
Thanks for your help.

closed 2018-03-27 by flounderscore