winder/Universal-G-Code-Sender #971: Error early in program not issuing feed-hold

winder, 2018-03-26:
## Problem description 
@breiler I noticed that UGS doesn't pause when an error is detected before the state has a chance to change from `IDLE` to `RUN`. I'm not sure if this is a new regression or an old one.

### Expected Behavior
UGS should issue a feed hold command even though the last reported state is `IDLE`.

### Actual Behavior
The program continues to run all commands up until the one which had an error, at which point it stops because the communicator correctly paused itself.

### Steps to Reproduce the Problem
Run the following program. Putting the error on the first line causes it every time, 
```gcode
F250.
G90 G20 G17
; attempt to fill up the buffer, adding/removing these lines
; causes the issue happens more/less frequently
; depending on the status poll rate setting
G1 X.123456789
G1 X-.123456789
G1 X.123456789
G1 X-.123456789
G1 X.123456789
G1 X-.123456789
G1 X.123456789
G1 X-.123456789
G1 X.123456789
G1 X-.123456789
G1 X.123456789
asdf
```

The strange thing is that it looks like this case is handled by this block in `GUIBackend.java`:
```java
    @Override
    public void pauseResume() throws Exception {
        logger.log(Level.INFO, "Pause/Resume");
        try {
            switch(getControlState()) {
                case COMM_IDLE:
                default:
                    if (!isSendingFile()) {
                        throw new Exception("Cannot pause while '" + getControlState() + "'.");
                    }
                    // Fall through if we're really sending a file.
                    // This can happen at the beginning of a stream when GRBL
                    // reports an error before we send it a status request.
                case COMM_SENDING:
                    this.controller.pauseStreaming();
                    return;
```

### Specifications
#### Version
master branch

#### Operating system
  Mac OSX

#### Platform
  GRBL 1.1f

breiler, 2018-03-26:
Went back as far as Jan 8 (2abdcdba6fed511d6a21662491e03342c6b3e4ad) and could reproduce the issue. 

I tried modifying `GrblController:rawResponseHandler` adding `this.pauseStreamingEvent();` which seems to do the trick:
```java
Optional<GcodeCommand> activeCommand = this.getActiveCommand();
if( activeCommand.isPresent() ) {
    processed =
        String.format(Localization.getString("controller.exception.sendError"),
            activeCommand.get().getCommandString(),
            lookupCode(response, false)).replaceAll("\\.\\.", "\\.");
            this.errorMessageForConsole(processed + "\n");
            this.commandComplete(processed);
            this.pauseStreamingEvent(); // Send a feed hold
        } else {
...
```

But this only work if there's support for streaming?

breiler, 2018-03-27:
I just realised that this would contradict the behaviour that is described in #970. The pause on error is used by @flounderscore to make a tool change. I guess that the work around if you want to us the errors this way would be to use single stepping mode until there is real support for tool changes?

winder, 2018-03-27:
@breiler true, UGS isn't good with tool changes, I don't think you can even fake it. GRBL won't wait until it gets to the tool change command, so it will most likely pause at the wrong spot. There are a couple solutions.

It would be trivial to make a gcode parser which looks for `M6` and sticks a dwell in front of it (or a custom routine even), I actually already have one that does this for `M3` commands: [M3Dweller.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/gcode/processors/M3Dweller.java)

It would also be cool to have a "pragma" feature which could tell the Controller/Communicator to do certain things. For instance throw in something like `(#ugs-directive FORCE_SINGLE_STEP)` and `(#ugs-directive RESTORE)` to have finer grained control over the program execution. That might allow UGS to do things like inserting breakpoints in your program ([like this grbl tool](https://github.com/duembeg/gsat)).

Another possibly simpler option might be to split the program up into separate blocks and use something like the `Workflow Helper` to send different sections.

flounderscore, 2018-03-27:
Are there any plans of implementing semi-automatic tool changing/probing like it's done in bCNC?

winder, 2018-03-27:
Resolved with #972

winder, 2018-03-27:
@flounderscore I think that would be a nice feature. For now, the recommendation is to have a separate gcode files for each toolpath.

closed 2018-03-27 by winder

winder, 2018-03-27:
@flounderscore the platform version of UGS has some automated probe routines, look in the plugin menu to open that module.