winder/Universal-G-Code-Sender #976: Plug-in Test probe

gelius59, 2018-03-28:
When the probe is working, the algorithm is
Move down with a great speed to touch
Move upward 1 mm
Move to the bottom with a slow speed at a distance set in the plugin.
The distance is the same as for high speed.
I think the distance, for low speed, should be configurable in the plug-in on the setup tab.
Or equal to 2 or 3 mm
Sorry for my bad english

winder, 2018-06-12:
Why would you need to configure the slow probe? Are you concerned that the plate will move between the fast and slow probe operations?

breiler, 2024-02-05:
There is a setting for the retract amount after the first probe:
[image]

closed 2024-02-05 by breiler