winder/Universal-G-Code-Sender #978: Add Units Preferance for Visualizer

Lizichka, 2018-03-30:
## Feature request for UGS Platform

I haven't found a way to switch Visualizer's units to inches. It's helpful to see the overall dimensions of the Gcode file before staring, but I have trouble using millimeters, the current units. I create the Gcode file in inches. 

Please add unit preference for Visualizer. Much appreciated.

dastultz, 2018-03-30:
Are you using GRBL? The display units come from the mode that GRBL is in.

breiler, 2018-08-07:
It now uses the preferred units setting which can be toggled in the jog-window.

closed 2018-08-07 by breiler

shoguntoo, 2018-09-05:
That is unfortunate, since one of my laptops is never connected to the cnc but use it to look at gcode and use the Visualizer