winder/Universal-G-Code-Sender #981: Probe Module Broke Bit

RecceDG, 2018-04-01:
Was running a job, used probe module to set Z height.

It came down, made initial contact, withdrew, slowly advanced, then stopped on recontact.

Remove probe, advance Z (Z-) the thickness of the probe plate, and zero out Z. Zero now set. 

(There has to be a way to set and use tool length offset, but I don't know what that is yet.)

Run job. Set up new job with different material thickness. Run probe module.

Z comes down, makes initial contact, retracts, then rapids full Z- trying to punch bit through probe. Bit snaps.

So worked fine once, then failed big.

jahnj0584, 2018-04-01:
Please post the macro you are using for probing

RecceDG, 2018-04-01:
I'm not - I'm using the probe module.

Whatever macro is the default, that's what I am using.

The only thing I changed was the thickness of my probe plate.

sckiffo2, 2018-04-09:
Then post console output what you have after accident.

closed 2018-05-17 by winder