winder/Universal-G-Code-Sender #983: Configure machine to have soft limits respected by jog in opposite direction from home.

RecceDG, 2018-04-01:
Related to https://github.com/winder/Universal-G-Code-Sender/issues/982

It should be possible to tell the machine the max travel limits in the opposite direction from machine home (where there are no limit switches, at least on an XCarve) so that when homing the machine sets soft limits on machine zero (per https://github.com/winder/Universal-G-Code-Sender/issues/982) it also sets soft limits on the other end of the machine, per configuration, than is respected by jog.

jahnj0584, 2018-04-01:
It already does that on my machine. Did you set the homing coordinate in
the proper corner?

breiler, 2018-04-01:
Can you post your firmware settings from the $$-command

RecceDG, 2018-04-01:
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 3    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 115    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 750.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 40.000    (X-axis travel resolution, step/mm)
$101 = 40.000    (Y-axis travel resolution, step/mm)
$102 = 188.976    (Z-axis travel resolution, step/mm)
$110 = 8000.000    (X-axis maximum rate, mm/min)
$111 = 8000.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 500.000    (X-axis acceleration, mm/sec^2)
$121 = 500.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$130 = 540.000    (X-axis maximum travel, millimeters)
$131 = 540.000    (Y-axis maximum travel, millimeters)
$132 = 100.000    (Z-axis maximum travel, millimeters)
ok

breiler, 2018-04-02:
You don't have soft limits enabled ($20), read this on how to activate:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean

This would resolve the #982 as well.

RecceDG, 2018-04-03:
Then there should be a UI element that reflects these status bits and allows the user to flip them as required.

carneeki, 2018-04-03:
The GRBL configuration can be edited by going to Machine -> Firmware
Settings, is this what you are after?

breiler, 2018-05-11:
There is now a setup wizard that should help setting up the machine. There is still some work to be done to configure the soft limits but the wizard should be enough to get those features working.

closed 2018-05-11 by breiler