winder/Universal-G-Code-Sender #987: visualizer sometimes does not act correctly

jahnj0584, 2018-04-03:
seems like its using an absolute command instead of relative, or vice versa. 

https://www.youtube.com/user/urgamecshk/videos?view_as=subscriber

breiler, 2018-04-03:
A bug report like this takes a lot of time to decipher so could you please provide more details when creating a issue. Otherwise it would probably take a lot of time for anyone to reproduce it...

I am guessing you are referring to this video:
https://www.youtube.com/watch?v=kVoO0kEfIbg

From the video I can see that the popup menu is old, what version of UGS are you running?
What version of GRBL are you running?
What is the output from `$$` and `$G`

jahnj0584, 2018-04-03:
1.1 and feb 18. I can't force reproduce it. Occasionally UGS will launch
and do this. This is the third time for me. It also doesn't send me macros
correctly when it is like this.

breiler, 2022-01-14:
Closing as the video is no longer available

closed 2022-01-14 by breiler