winder/Universal-G-Code-Sender #989: Send status panel problems in UGS Platform

breiler, 2018-04-04:
## Problem description 

There are a couple of small issues with the send progress panel that I've started working on. But the code changes grew out of control, so this issue is created to have them organised and to discuss the solutions.

### Behavior
* [ ] When pausing a job the **Duration** timer doesn't stop in the status panel. I would expect the duration timer to be paused/resumed together with the job.
[image]
* [ ] When the job is finished the **Sent row** count is wrong. I expect that the number of commands in the file is the same number processed.
[image]
* [ ] When the panel is made wider it can't be shrunk again.

### Specifications
#### Version

UGS Platform 2.0 - development branch

#### Operating system

Mac OSX

#### Platform

GRBL 1.1f

#### Other

breiler, 2018-04-04:
The problem with the wrong number of sent rows exists because of two problems:
1. It counts sent rows in the Controller not rows in processed in the stream.
1. It stops the status panel update timer before all commands has finished because the finished file event is triggered when the send buffer is empty in `AbstractController.java`
```java
    public void checkStreamFinished() {
        if (this.isStreaming() && !this.comm.areActiveCommands() && this.comm.numActiveCommands() == 0) {
            String streamName = "queued commands";
            boolean isSuccess = (this.errorCount == 0);
            this.fileStreamComplete(streamName, isSuccess);

            // Make sure the GUI gets updated when the file finishes
            this.dispatchStateChange(getControlState());
        }
    }
```

What do you think of using fetching the values from streamCommands instead of trying to store command counters in the AbstractController:
```java
    public int getRowStat(RowStat stat) {
        switch (stat) {
            case TOTAL_ROWS:
                return this.numCommands;
                //return streamCommands.getNumRows(); <-- reactivate this part of the code
            case ROWS_SENT:
                return this.numCommandsSent;
                //return streamCommands.getNumRows() - streamCommands.getNumRowsRemaining(); <-- reactivate this part of the code
            case ROWS_REMAINING:
                return this.numCommands - this.numCommandsCompleted - this.numCommandsSkipped;
                //return streamCommands.getNumRowsRemaining(); <-- reactivate this part of the code
            default:
                throw new IllegalStateException("This should be impossible - RowStat default case.");
        }
    }
```

Then the `AbstractController.queuedCommands` would be obsolete as it's only used by test cases.

winder, 2018-04-06:
I think the reason I did it this way is that the preprocessor can add/remove commands. In the visualizer, I tried to show the status based on the original code. If the command was a big arc it could have been split into 50 line segments. The GcodeStream reader/writer stores the original line number in addition to the new line number.

I'm not sure what is better. Would it be weird to open a file with 100 lines and see 1000 lines in the GUI?

breiler, 2018-04-06:
Yeah, that would probably be weird... 
Sorry, I didn't thought of that, I'll keep digging.

breiler, 2018-04-09:
There is a couple of test cases that I've working with.

**Scenario 1: Stopping a running file**
   1. Open and connect UGS
   1. Open a file (ex. stress_test.gcode) and run it 
   1. Stop the file
   1. Jog a couple of times and the remaining rows decreses
   [image]

**Scenario 2: Running a small file**
  1. Open and connect UGS
  1. Open a small file (ex. tiny.nc), run it and let it complete
  1. It doesn't how many rows was processed:
  [image]

**Scenario 3: Running a large file**
   1. Open and connect UGS
   1. Open a file (ex. stress_test.gcode), run it and let it complete
   1. It is not counting the last couple of rows
   [image]

**Scenario 4: Running a file with comments**
   1. Open and connect UGS
   1. Open a file (ex. buffer_stress_test.gcode), run it and let it complete
   1. It is not counting the comments (there are 52 command rows in the file)
   [image]

**Scenario 5: Running a file in workflow helper**
  1. Open and connect UGS
  1. Open the workflow helper
  1. Add `buffer_stress_test.gcode` and `tiny.nc`
  1. Run the first file, and a dialog is shown saying the file is finished but it's still running commands in the background.
  [image]