winder/Universal-G-Code-Sender #993: Workflow Helper Plugin Finished Check Box Not Working

kenfritz, 2018-04-07:
Problem description:
The latest releases of UGS Platform (Build #997) the Workflow Helper Plugin has stopped showing the Checked box under the Finished Column after a file has completed being carved. Also, the Workflow Event Popup box no longer appears after a completed carve. This function was working as of Build #979.

Here is what the Workflow Helper looks like after the toolpath file is run now.
[image]

Expected Behavior:
Below capture shows what Build# 979 showed after the toolpath is run. Finished column showed Checked box and Popup workflow message.
[image]

Steps to Reproduce the Problem
  1. Open Workflow Helper
  2. Open toolpath file
  3. Run toolpath
  4. Notice no check after completed toolpath is run
 
 Version:
 - UGS Platform 2.0 - Nightly Build 997 Not working
-  UGS Platform 2.0 - Nightly Build 979 working
 
Operating system:
  Windows 10

Platform:
X-carve
 GRBL 1.1f

breiler, 2018-04-07:
I think I've introduced the problem with this commit: a411cee2fd81401a6ac08eaabdc0f32d12513602
The method `GUIBackend. isSendingFile()` is reporting that the file is finished sending before the comm-state is changing to COMM_IDLE. This will cause the workflow helper to never detect if a file is completed:

WorkflowWindowTopComponent.UGSEvent:
```java
        if (cse.isStateChangeEvent()) {
            if (wasSending && cse.getControlState() == ControlState.COMM_IDLE)
               this.completeFile(backend.getGcodeFile());
            wasSending = backend.isSendingFile();
        }
```

I'll have a look at it!

breiler, 2018-04-07:
If this fix is accepted this would fix your issue but introduce another. =/

It will report the file as completed when all commands have been sent to the controller, not when all commands have been finished processing. 

So a dialog would appear with the file complete message but the file would still be running in the background. I will attempt to fix this in issue #989.

[image]