winder/Universal-G-Code-Sender #996: Two instances of UGS.

BallscrewBob, 2018-04-07:
## Feature request
Ability to run two instances of UGS v2.0 nightly
Built a second machine but if I try open a new instance it just re-opens the existing instance.
Is there a method to do this easily ?

winder, 2018-04-09:
There are some directions for how to do this here, let me know if that doesn't work and I can try playing around with the settings:
http://wiki.netbeans.org/FaqMultipleIdeInstances

closed 2018-04-09 by winder

BallscrewBob, 2018-04-10:
> @BallscrewBob <https://github.com/BallscrewBob> which part are you stuck
> on? I gave it a try and the application seems to launch alright:
> [image: screen shot 2018-04-10 at 8 45 08 am]
> <https://user-images.githubusercontent.com/125509/38557631-9ed1ca30-3c9b-11e8-9bdf-955a3298bfec.png>

NICE !...I dont use Mac';s but thats a great help...I can script that into
a new shortcut in Windows.
Will let you know how it works and thanks for the help.
Much appreciated.

winder, 2018-04-10:
The link I shared earlier has the windows command as well, it looks very similar but also includes a cache directory. That seems odd to me, maybe it's optional. You should be able to get it working though.