winder/Universal-G-Code-Sender #998: UGS "ghost" jogs

jahnj0584, 2018-04-08:
### Expected Behavior
Sometimes, if i hold down jog commands they all execute in real time. Once i let go, the machine stops moving

### Actual Behavior
Other times, randomly, the machine will continue to move afterwards. Its like on certain days UGS is taking longer to buffer or send the commands?

### Steps to Reproduce the Problem
hold down CTRL + Arrow key to jog an axis at 5000mm/min and 1mm step size

### Specifications
Feb 18 nightly 
#### Operating system
W 7

#### Platform
1.1

winder, 2018-04-08:
"Sticky Keys" is an operating system feature, you should be able to observe the same behavior in a text editor. Hopefully some combination of #779 and #947 can solve this in the future.

closed 2018-04-08 by winder

jahnj0584, 2018-04-08:
No that's not what's happening. Simply restarting UGS can solve the issue.

terjeio, 2018-05-01:
@jahnj0584 : are you running on a Mega? If so you may try my modifed [grbl,](https://github.com/terjeio/grbl-Mega) I believe the jog cancel in the master is badly implemented - perhaps this is better? I have implemented 3-key rollover cursor-key jogging in my gcode-sender (not public) and so far I have not managed to fail jogging even when abusing the keys (touch wood) - this with my ARM port of grbl which has similar changes to jog cancel.

jahnj0584, 2018-05-03:
Uno