winder/Universal-G-Code-Sender #1001: Z Step size permanently disabled

igorwapenik, 2018-04-09:
Hi! I am having the following issue, the Z index step size box is permanently disabled and I cant find an way to enable it.  

I have the same software version running in another computer and I cant simulate the issue there.

Image as how the input stays:
[image]

breiler, 2018-04-09:
You can either use the menu, or the preferences "UGS -> Sender Options -> Use separate step sizes...":

[image]

igorwapenik, 2018-04-09:
I tried that already, as you can see it is also disabled:
[image]

breiler, 2018-04-09:
What version of GRBL are you running?

breiler, 2018-04-09:
Never mind, I just reproduced it. It occurs when connecting for the first time. I have a fix for it soon.

igorwapenik, 2018-04-09:
The strage part is that I am running this version on two notebooks, one is presenting this problem and the other doesn't.

breiler, 2018-04-09:
This should be fixed in the next nightly build. 
As a work around, connect -> disconnect and then connect again this should make the menu option available.

igorwapenik, 2018-04-09:
Thanks for the help! By connecting and disconnecting I was able to edit the Z Step size. Thanks!