winder/Universal-G-Code-Sender #1005: Probe Module Plugin Bug - Retract Amount movement using Inches instead of mm

kenfritz, 2018-04-11:
## Problem description 
Probe Module Plugin Retract Amount movement is using Inches instead of mm. This is causing the soft limit alarm message (See attachment 1 - ACTUAL BEHAVIOR). When I remove my soft limits on the Z-axis it crashed my spindle at the top of the Z-Axis. Ouch. This bug is seen in the latest build #1005. I do not see this problem in build #986 and earlier.

Note: Probe Module Broke Bit #981 reported bug similar with different failure method but could be same issue.

### Expected Behavior
All my settings for Probe Module are in mm. When the probe is retracted the distance traveled is approximately 13mm (Retract Amount (10mm) plus Plate Thickness (3.023mm). I have been running it like this for months with no problem.

[image]

Working version: Build #986 (Console Output)
ok
>>> G21 G91 G49
>>> G38.2 Z-10 F125
>>> G91 G20
ok
[PRB:-30.8662,-31.2599,-1.0996:1]
ok
ok
>>> G91 G21 G0 Z1.0
>>> G91 G21
ok
ok
>>> G21 G91 G49
>>> G38.2 Z-10 F30
>>> G91 G21
ok
[PRB:-30.8662,-31.2599,-1.1002:1]
ok
ok
>>> G10 L20 P1 Z3.023
ok
ok                        THIS IS THE RETRACT GCODE COMMANDS THAT WORK CORRECTLY
>>> G91 G21
>>> G90 G21 G0 Z13.023
>>> G91 G21
ok

### Actual Behavior
This is the probe module not working.

NOT Working Version: Build #1005 (Console Output)
ok
>>> G21 G91 G49
>>> G38.2 Z-10 F125
>>> G91 G20
>>> G91 G21 G0 Z1.0
>>> G91 G20
>>> G21 G91 G49
>>> G38.2 Z-10 F30
>>> G91 G20
ok
[PRB:-30.8662,-31.2599,-1.1015:1]
ok
ok
[PRB:-30.8662,-31.2599,-1.1013:1]
ok
ok
>>> G10 L20 P1 Z-2.9111000000000002
ok
              THIS IS THE RETRACT GCODE COMMANDS AND WHERE THE PROBLEM IS I THINK. SEE 
              WORKING VERSION ABOVE.
ok
>>> G91 G20
>>> G90 G21 G0 Z13.023
>>> G91 G20
ok
An error was detected while sending 'G90 G21 G0 Z13.023': (ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked. Streaming has been paused. [MSG:Reset to continue]

Attachment #1:

[image]

### Steps to Reproduce the Problem

kenfritz, 2018-05-17:
Will
I tested build #1031 today after the fix to the Probe Module. My tests showed absolutely no change. I am getting the same probe issued that I explained with Case #1005. Can you review the fix that applied to the latest build?
thanks.

    On Thursday, May 17, 2018, 6:48:07 AM PDT, Will Winder <notifications@github.com> wrote:  Closed #1005.

winder, 2018-05-18:
@kenfritz I fixed a bug in the Z probe routine causing the WCS offset to be set incorrectly which in turn would cause the retract to be too large. It looked related to your bug report since the very thorough description (thank you) showed that the offset was being set incorrectly and then a soft limit is triggered it seemed pretty related. I'll give it another test in the morning.

winder, 2018-05-18:
@kenfritz it looks like there are some stale files, so the new code isn't loading automatically. I just made another change that should hopefully clean this up.