winder/Universal-G-Code-Sender #1008: Send UI button grayed after loading a file for the first time

BusbyGT, 2018-04-11:
## Problem description 
When loading a Gcode file for the first time, the UI button to start/send the code is grayed out after it loads. You have to go to the Machine menu and start the job there. After the first file, every other file works okay. This is UGS Platform 64-bit on Windows 10 and the nightly build from maybe a week ago. I’ll get the actual version later today.

### Steps to Reproduce the Problem
  1. Open the program
  2. Click on the toolbar menu icon for opening a file
  3. Selects a file in the dialog and click ok
  4. The file is loaded into Visualizer and no errors are thrown but the Send/Start button remains grayed out. The menu option is available and must be used.

### Specifications
#### 
  
  UGS Platform 2.0 - Nightly Build Xx/XX/xxxx

#### Operating System
  Windows 10

#### Platform

GRBL v1.1f

breiler, 2018-04-11:
This should be fixed in #1003.

BusbyGT, 2018-04-11:
I’ll pull a new daily build this afternoon and see if it’s corrected.

Thanks!

closed 2018-04-13 by winder