winder/Universal-G-Code-Sender #1009: Setup wizard

breiler, 2018-04-15:
## Feature request
I would like to see a setup wizard that would make it easier to get started with UGS configure a controller. 

I've started the implementation on a minimum viable product and would like your input. 

* Connecting to GRBL:
[image]
[image]
* Motor wiring
[image]
* Calibrating step lengths
[image]
* Configuring limit switches
[image]
* Configuring homing
[image]
* Configuring soft limits
[image]

vMeph, 2018-04-15:
great job!!!

RecceDG, 2018-04-16:
I'd like to see a UI button that is "set soft limit on [X|Y|Z] axis here"

breiler, 2018-04-16:
@RecceDG you mean a button somewhere more accessible than in the last step of a setup wizard, like in the toolbar menu?

There is a problem that, in order to use soft limits you need to have homing enabled (assuming you are using a GRBL based controller). And now we are just talking about enable/disable. There is a lot of other parameters that should be available eventually; direction, debounce, pull-off, seek rate and feed rate.

A prerequisite to enabling homing is that you need hard limits enabled.

So there is a lot of stuff that needs to be configured before being able to enabling soft limits. 
How should such a button behave if the requirements for enabling soft limits isn't met?

jahnj0584, 2018-04-16:
I think he means like use current machine POS as the soft limit length.

I just jog until i hit my hard stops. Whatever that value is i go 1mm less

RecceDG, 2018-04-16:
> A prerequisite to enabling homing is that you need hard limits enabled.

That isn't the case. My machine will happily crash the limit stops.

I have limit switches on X, Y, X minimums. First thing I do on startup is home the machine to set machine zero. That works great. But currently, the machine does nothing with limits, hard or soft.

Yes, I can manually set them via the $ codes - I just haven't done it yet. But really, these should be UI elements. There should be a light "hard limits engaged", "soft limits engaged". If you try and jog past a soft limit, it should jog up to the soft limit, then light "[Z|Y|Y] Limit" and maybe sound a tone?

And yes, it would be helpful to be able to set current machine position to either min or max soft limit (so you can protect fixtures, etc) plus be able to *clear* soft limits (which should reset the mins to machine zero, and the max to the configuration value in your wizard.

I'm OK with soft limits being conditional on homing.

I've been doing a lot of complicated carves with my XCarve lately, and I keep tripping on UGS UI and usability problems. I keep meaning to do a quick MSPaint mockup of what I think needs to be on the UI.

breiler, 2018-04-16:
@RecceDG in the issue #983 you had neither hard limits ($21) or soft limits ($20) enabled. Have you changed your configuration?

@jahnj0584 setting software limits by jogging to that location is a good idea, I might add it later.

breiler, 2018-05-03:
The setup wizard is now available in the latest nightly build. Please give it a try and any feedback is appreciated.

This is first version and there's room for a lot of improvements.  I have a backlog with stuff that I want to change or add. But it would be nice to know if it works this far! =)

breiler, 2018-05-11:
I tweaked the timeout.
The wizard is in place and it looks like it's working.

closed 2018-05-11 by breiler