winder/Universal-G-Code-Sender #1017: An unexpected command was completed by the controller. after grbl1.1f update

Cabalist, 2018-04-27:
## Problem description 

Sometimes when connecting to a woodpecker board running grbl1.1f I get the following error:

```Error while processing response <ok>: An unexpected command was completed by the controller.```

*The machine appears to work fine otherwise.*  I was not getting this error before upgrading grbl1.1f.  Sometimes it connects fine but disconnecting and reconnecting a few times will cause this error to appear.

### Expected Behavior

Not getting an error message.

### Actual Behavior
[image]

### Steps to Reproduce the Problem
  1. Open the program
  1. Click on the serial connect button
  1. The error message appears

### Specifications
#### Version

Product Version: Universal Gcode Platform 201609300101
Java: 1.8.0_121; Java HotSpot(TM) 64-Bit Server VM 25.121-b13
Runtime: Java(TM) SE Runtime Environment 1.8.0_121-b13
System: Mac OS X version 10.13.4 running on x86_64; UTF-8; en_US (ugsplatform)
User directory: /Users/work/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev
Cache directory: /Users/work/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev/var/cache

#### Operating system
Mac OS X version 10.13.4 

#### Platform
  GRBL 1.1f

1610 CNC from china

#### Other
Terminal output: 
```
**** Connected to /dev/cu.wchusbserial1410 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 5    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 800.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
**** Connection closed ****
15,128|FS:0,0>
**** Connected to /dev/cu.wchusbserial1410 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 5    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 800.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> $G
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 5    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 800.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
Error while processing response <ok>: An unexpected command was completed by the controller.
```

breiler, 2018-04-29:
I'm not able to reproduce this error with the nightly build.
It seems that you have an old version of UGS (20160930), could you try the latest nightly build to see if the problem still exists?

Cabalist, 2018-04-29:
I have downloaded the latest builds UGS Platform from the README in github.   There are two links but I believe I have the latest one (i.e. non stable).

UGS Platform - Universal Gcode Platform, the next generation platform-based interface
--

http://bit.ly/1DkClRW
https://winder.ci.cloudbees.com/job/UGS/lastSuccessfulBuild/artifact/UGSPlatform/dist/ugsplatform.zip

Gives me:
```
Product Version: Universal Gcode Platform 201609300101
Java: 1.8.0_121; Java HotSpot(TM) 64-Bit Server VM 25.121-b13
Runtime: Java(TM) SE Runtime Environment 1.8.0_121-b13
System: Mac OS X version 10.13.4 running on x86_64; UTF-8; en_US (ugsplatform)
User directory: /Users/work/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev
Cache directory: /Users/work/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev/var/cache
```

As I am typing this I noticed in the titlebar it is different:
[image]

UGS Platform - Somewhat stable build Universal Gcode Platform, the next generation platform-based interface (Requires Java 8)
--

http://bit.ly/2rk84y2
https://raw.githubusercontent.com/winder/builds/master/UniversalGCodeSender/ugsplatform-2.0.0-stable.zip

Gives me:
```
Product Version: NetBeans Platform 8.1 (Build 201510222201)
Java: 1.8.0_121; Java HotSpot(TM) 64-Bit Server VM 25.121-b13
Runtime: Java(TM) SE Runtime Environment 1.8.0_121-b13
System: Mac OS X version 10.13.4 running on x86_64; UTF-8; en_US (serverlaunch)
User directory: /Users/work/Library/Application Support/ugsplatform/dev
Cache directory: /Users/work/Library/Application Support/ugsplatform/dev/var/cache
```

[image]

breiler, 2018-04-29:
Oh, I'm sorry! The version string isn't updated correctly in the about dialog.

ukoksoy, 2018-06-13:
I have the same problem. 
It may be related with the Arc Setting. I was having problems with G2/G3 commands in the previous versions, so I choose from settings to change G2/G3 commands to G1.
I installed the latest version today. I imported gcode file. Although there are 42000 rows in the file, it recognizes only 1450 row. Here is the gcode command on the 1450th row:
G19 G3 X42.3957 Y15.587 Z-4.9025 J0.0168 K0.042

When I go back to previous Arc Setting which enables G2/G3 commands, this time all the 42000 rows are recognized, However, the program gave below error in 1669th row.

I attached gcode file as well.

[image]

[gcode file.zip](https://github.com/winder/Universal-G-Code-Sender/files/2098862/gcode.file.zip)

toffuser, 2019-03-29:
I have the same problem just by trying to HOME the machine

JacksCrafts, 2019-03-29:
Could it be your bed size settings?

JacksCrafts, 2019-03-29:
I do not use Homing myself.
But you might check to make sure it is on.
$$
check $22 in on
and $130 $131 $132 are the limits of your bed size.

Saur0o0n, 2020-02-22:
Hi,
I have the same problem with nigtly build. On Candle (for example) - there is no problem. Also program is not stopping all the time in the same place (if repeated few times).

Error: "An unexpected command was completed by the controller"
```
com.willwinder.universalgcodesender.AbstractController$UnexpectedCommand: An unexpected command was completed by the controller.
	at com.willwinder.universalgcodesender.AbstractController.commandComplete(AbstractController.java:793)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:158)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:815)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:172)
	at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:194)
	at java.lang.Thread.run(Thread.java:748)
```

Console output:
```
ok
>>> X-75.341Y-13.148
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
**** Connected to ttyUSB0 @ 57600 baud ****
Grbl 1.1g ['$' for help]
```

breiler, 2020-12-06:
Looking through the logs it looks like there are unstable connection with the controller. The welcome message is displayed in the middle of the log which indicates that the controller has been reset. Check your USB cable and make sure it is a good quality shielded one.

If there still problems please open a new fresh issue and we'll take it from there.

closed 2020-12-06 by breiler

Saur0o0n, 2020-12-06:
Actually after some trail and errors I've assumed its because of system load. This may be false lead, but (it's not dedicated system to cnc machine only) when the system was doing other cpu intensive things this errors where often. I've tried to bump the priority to even -19 - but it wouldn't help. Perhaps some java stuff like GC or something else is here to blame... well it's java :/
I guess Candle handles simply other way pushing data - in this case it work's better.

Yasir898, 2023-02-09:
i face a problem
pen plotter is working fine like stepper motor and servo
but after a some execution it stack and pause and show the error of grbl 1.1f and i have to do it again

**** Connected to COM3 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> G21G90 G0Z5
>>> G90 G0 X0 Y0
>>> G90 G0 Z0
ok
ok
ok
>>> $J=G21G91X10Y10F10000
ok
**** Connected to COM3 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
[... 538 lines cut ...]
>>> G1X73.0957Y77.4824
>>> G4P0.2
>>> M05S0
>>> G4P0.2
>>> G1F10000
>>> G2X68.6787Y80.2104I1.6554J7.6204
ok
ok
ok
>>> G2X66.976Y85.0781I5.9919J4.8276
ok
ok
ok
>>> G2X67.5855Y88.1355I8.187J-0.0428
ok
Grbl 1.1f ['$' for help]

**** Canceling file transfer. ****

>>> $J=G21G91X10Y10F10000
ok
errr:7errr:7
An unexpected error was detected: 

[image]

breiler, 2023-02-09:
Please create a new issue @Yasir898 with all details.