winder/Universal-G-Code-Sender #1018: Unexpected behaviour of G28|G30 after using "return to zero" button

RecceDG, 2018-04-28:
OK, so here’s a weird one.

Using Universal GCode Sender platform nightly build:

1.  Move carriage to YMax, X roughly centred. Z all the way up. Enter G28.1 on the command line. This saves the position in what I’m calling the “clean and remove workpiece” position.

2.     Move carriage to YMin, X roughly centred, Z all the way up. Enter G30.1 on the command line. This saves the position in what I call the “tool change position”.
 
3.     Type G28 on command line. Machine rapids to the proper position.
 
4.     Type G30 on the command line. Machine rapids to proper position.
 
5.     Jog carriage to arbitrary point, then zero X, Y, Z (to set work zero).
 
6.     Type G28 on command line. Machine rapids to proper position.
 
7.     Type G30 on command line. Machine rapids to proper position.

8.     Load GCode created with Fusion 360 that has “safe G28” post processor option active. Run program. Program runs normally (using part zero set in step 5 properly) then on completion of run, lifts Z to G28 then rapids back to G28. (Program issues G28 Z0; G28 X0 Y0;)

9.     Type G30 on command line (actually, execute stored macro G30 Z0; G30 X0 Y0) and machine rapids to G30 position.
 
10.     Hit “return to zero” button on UGS (issues G90 X0 Y0; G90 Z0) machine rapids back to part zero.

OK, so far everything is working exactly the way I expect. Now it gets WEIRD.

11. Jog machine to arbitrary location.

12.  Type either G28 or G30 on command line.

13.   Z axis lowers to part zero, then retracts to G28|G30 zero, then rapids X/Y to part zero, then rapids back to G28|G30 zero. Every use of G28 or G30 from this step on will ALWAYS first move to part zero before executing the move to either the G28 or G30 positions.

What. The. Hell.

It seems like G90 is somehow setting a switch somewhere that forces subsequent G28 and G30 moves to first move to part zero, which is SUPER NON INTUITIVE.

Is this working as designed? If so, how do you clear it? Or is this a GBRL bug?

breiler, 2018-04-29:
I wasn't able to reproduce this. What version of GRBL are you using?
Could you post the output from the commands: `$$`, `$#` and `$N`. 
When you are able to reproduce the problem it would be interesting to see the output from `$G`.

RecceDG, 2018-04-29:
>>> $H
ok
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 3    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 115    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 750.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 40.000    (X-axis travel resolution, step/mm)
$101 = 40.000    (Y-axis travel resolution, step/mm)
$102 = 188.976    (Z-axis travel resolution, step/mm)
$110 = 8000.000    (X-axis maximum rate, mm/min)
$111 = 8000.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 500.000    (X-axis acceleration, mm/sec^2)
$121 = 500.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$130 = 520.000    (X-axis maximum travel, millimeters)
$131 = 482.000    (Y-axis maximum travel, millimeters)
$132 = 100.000    (Z-axis maximum travel, millimeters)
ok
>>> $#
[G54:-468.200,-430.200,-59.420]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:-265.000,-18.720,-1.000]
[G30:-417.400,-481.000,-1.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
>>> $N
$N0=
$N1=
ok
>>> G28
ok

*Worked*

>>> G30
ok

>>> $J=G20G91Z-0.1F40
ok
>>> $J=G20G91Z-0.1F40
ok
>>> $J=G20G91Z-0.1F40
ok
>>> $J=G20G91Z-0.1F40
ok
>>> $J=G20G91Z-0.1F40
ok
>>> $J=G20G91X0.1F40
ok
>>> $J=G20G91X0.1F40
ok
>>> $J=G20G91X0.1F40
ok
>>> $J=G20G91X0.1F40
ok
>>> $J=G20G91X0.1F40
ok

*Jogging*

>>> g30
ok

*Worked*

>>> $J=G20G91Z-1F40
ok
>>> $J=G20G91X1F40
ok
>>> $J=G20G91X1F40
ok
>>> $J=G20G91X1F40
ok
>>> $J=G20G91Y1F40
ok

*Jogging*

>>> G30 Z0
ok
>>> G30 X0 Y0
ok

*Lowered Z to zero, then raised to G30 height, then moved to XY zero, then moved to G30 XY*

>>> g28
ok

*Worked*
>>> $J=G20G91Z-1F40
ok
>>> $J=G20G91Z-1F40
ok
>>> $J=G20G91X-1F40
ok
>>> $J=G20G91X-1F40
ok
>>> $J=G20G91Y-1F40
ok
>>> $J=G20G91Y-1F40
ok
>>> $J=G20G91Y-1F40
ok

*jogging*

>>> G28 Z0
ok

*Lowered Z to zero, then raised to G28 height*

>>> G30
ok

*worked*

OK, so this looks like it is related to the moving of individual axis - and might be related to the Z0, which my reference stated was a way to select an axis, but now looks like is commanding a move to zero before moving to G28 or G30, which makes sense.

AHA! I found another reference that states you need to use a G91 first.

>>> $J=G20G91Z-1F40
ok
>>> $J=G20G91Z-1F40
ok
>>> $J=G20G91X1F40
ok
>>> $J=G20G91X1F40
ok
>>> $J=G20G91X1F40
ok
>>> g91 G30 Z0
ok

*WORKED*

OK, this is Working as Designed. Im a victim of an (outdated) reference.

closed 2018-04-29 by RecceDG