winder/Universal-G-Code-Sender #1022: "GRBL has not finished booting." issue after the windows update

mateuszbugaj, 2018-05-03:
Hi, some time ago i build a cnc based on a grbl that was running by UGS and it worked without a problem for few months. Two days ago there was some Windows 10 update and my guess is that this caused my problems.

[image]

I'm using Arduino UNO (clone but without CH340) with CNC shield (GRBL 0.8c compatible as Ebay description says) and UGS v2 [nightly]

I tried to use diferent versions of UGS like version 1.0.9 or Universal Gcode Platform. I also changed Arduino Uno and reloaded grblUpload sketch. I instaled java actualisation.
I read some similar issues on this forum but my setup worked and sudenlly stoped while usualy those issues are posted by people who have problem with new setup. I would be happy to see your suggestions. Thanks

winder, 2018-05-03:
Some common causes and steps to try for this issue are described here:
http://winder.github.io/ugs_website/guide/troubleshooting/#grbl-has-not-finished-booting

mateuszbugaj, 2018-05-03:
I already tried these, also grbl works when I'm using Arduino IDE

winder, 2018-05-03:
What version of GRBL are you using? The 9600 baud rate hasn't been used in a very long time.

mateuszbugaj, 2018-05-03:
V0.9, worked well with 9600 baud rate till now

winder, 2018-05-03:
Have you tried 115200? That is the default setting for v0.9.

mateuszbugaj, 2018-05-03:
It's working! It was set by default to 9600 and i didn't change it. I had no idea it should be 115200 because i never had any problems with that. Any way, I am very grateful, thanks, Mr. Winder.

closed 2018-05-03 by winder