winder/Universal-G-Code-Sender #1025: Welcome Page

winder, 2018-05-06:
## Feature request
An in app landing page with some documentation to direct new users to various features they may find useful but may not discover on their own. It should be displayed in the same box as the visualizer and have focus by default and have a "show on startup" checkbox to optionally disable it in the future.

## Problem description 
More and more killer features are being added which aren't easy to discover: setup wizard, advanced jogging module, pendant UI, macro preferences, probe module, etc. Some sort of welcome page which pops up over the visualizer window could be a good way to help with discovery.

The technical part is pretty simple, and Netbeans has a [WelcomeComponent](https://github.com/apache/incubator-netbeans/blob/master/welcome/src/org/netbeans/modules/welcome/WelcomeComponent.java) if any additional inspiration is needed. Mainly I'm imagining some static content to inform users about features and workflows. Some ideas:

* Spotlight for important features like setup wizard, probe module, pendant UI, macro preferences. If there are enough of these, maybe have them be randomized in a tip-of-the-day style.
* Links to various online resources to help users find out where to learn more, ask questions, report bugs, etc. For example: UGS website, UGS google group for Q&A, Github issue page, gcode documentation, grbl documentation,...)
* Information about localizing UGS.
* Multiple tabs/buttons to load key tutorials (Could have a `WelcomeTutorial` class for Lookup extension):
    * For first time users a guided how to connect, load a file and send it might be useful. This could even interactively guide you through connecting, and loading/sending a file similar to how the Setup Wizard guides the user through the connection.
    * How to install / configure the "[darkula](http://plugins.netbeans.org/plugin/62424/darcula-laf-for-netbeans)" theme.

## Suggestions Wanted
If anyone has suggestions specific suggetsions about how this page could look in order to be an engaging and inviting first impression to the UGS Platform I'd be curious to hear them. @dastultz I'm especially curious if you have had any ideas about this.

Also any suggestions for content that would be most useful.

dastultz, 2018-05-08:
I use IntelliJ IDEA daily. I like the "tips and tricks" box that opens on startup. Perhaps the build process could generate a semi-random sequence of tips starting with new features first.

The idea of a welcome screen "tab" on the visualizer/console pane is good, too. Perhaps if the documentation on the website was improved/restructured you could just embed a browser in that pane with the table of contents.

I really like the popup documentation in Fusion 360, hover over a checkbox you don't understand and a description, usually with an illustration, comes up. Looks pretty challenging to maintain.

Do you i18n you web site, too?

carneeki, 2018-05-08:
Fusion 360's hover-docs really are the nicest I think I've ever seen in
software. If UGS wanted to replicate it, I would definitely be in favour of
that, but I imagine that's probably a huge effort and I'd be understanding
if it took a while.

winder, 2018-05-13:
I've been messing around with the Netbeans Welcome component and am now starting to work on content. I haven't modified the style much beyond removing the Netbeans icons, so the style is still subject to change. So far the tabs I'm considering are:

**Getting Started**: Focus for first run, tells users the absolute minimum needed to get started using UGS assuming their machine is already configured (maybe a link to the configuration wizard would be good in case the machine is not configured).

**Recent**: A list of recent directories/files which have been used.

**Features**: Spotlight section introducing users to things like the Window/Plugin Menu, Keybindings, etc.

**What's New**: Dynamic page loaded from content in github to introduce new features/content. Netbeans uses an RSS feed for this, but I think a simple template would be more suitable for UGS. It could have an indicator if you need to upgrade to get the feature.

Here's a little demo of where it's at right now:
[image]

winder, 2018-05-15:
I'm on a roll tonight, here is a first-pass of a features tab. I think the **What's New** section can use the same layout:
[image]

dastultz, 2018-05-16:
Looking good! I'm mentally preparing for my role in this...

carneeki, 2018-05-16:
Smashing work Will!

winder, 2018-05-16:
If anyone wants to help with the content/copy this is the format I'm using, I can plug in as many of these as we want (or even randomize/cycle through them if there is enough content). The information needed for a feature looks like this:

feature name: "Digital Read-Out"
description: "The DRO is your first stop for the current ..."
image: 300px (HiDPI), 150px tall (standard)

This is what I have so far (subject to change, I wrote it up quickly to get some content to test the rendering):
```
"Plugin Support!"
"Most plugins can be found in the 'Window > Plugins' menu. New plugins are added all the time, see the what's new section for the latest features!"
"com/willwinder/ugp/resources/features/new_plugins.png"

"Digital Read-Out (DRO)"
"The DRO is your first stop for the current machine status. It tells you the work/machine coordinates, machine/spindle speeds, gcode state and more! It can be used to reset
individual axes by clicking on the axis label, and has dynamic work position controls by clicking on the coordinate numbers."
"com/willwinder/ugp/resources/features/dro.png"

"Jog Controller"
"The Jog Controller is your primary tool for manually controlling your machines location. It has the option of using separate step sizes for XY and Z axes in addition to a c
lick-and-hold mode for continuous jogging."
"com/willwinder/ugp/resources/features/jog_controller.png"

"Custom Macros"
"User defined macros can be configured in the preferences menu. A variety of convenient substitutions are available to help create perfectly tailored time savers for your wo
rkflow. See the help menu for more details."
"com/willwinder/ugp/resources/features/custom_macros.png"

"Configurable Keybindings"
"Nearly every feature in UGS can have a configurable keybinding. From the preferences menu open the keybinding section for a complete list of actions which can be configured
 (including your custom macros!)."
"com/willwinder/ugp/resources/features/keybinding.png"

"Probe Module"
"The probe module can be opened from the plugins menu, it provides first class support for common probe devices. Some of the supported routines include standard Z-Depth prob
ing and 3-Axis corner probing."
"com/willwinder/ugp/resources/features/probe_module.png"
```

breiler, 2018-05-22:
My english isn't the best, so please feel free to correct! =)

---
```
"Setup wizard"
"Configure and calibrate your hardware with the Setup wizard. It will help you to setup and test limit switches, step length, homing and soft limits. Start the wizard from the machine menu."
```
[image]
[image]

---
```
"Workflow helper"
"Manage complex projects with multiple files using the Workflow helper. It will keep track of which files that has been run and what tool to use for the next file."
```
[image]
[image]

winder, 2018-05-23:
Made some progress on the **Recent Work** tab
[image]

breiler, 2022-12-15:
Closing as this is merged and fully functional.

closed 2022-12-15 by breiler