winder/Universal-G-Code-Sender #1031: 1022 update errors

vMeph, 2018-05-12:
## Problem description 
after reinstall ugs #1022 update
when i open UGS Platform, om start i have 2 visualizer windows and program crashes,
I add to go to %appdata% and  delete .ugsplatform folder
when i open now is all back to normal but now the jog controller is back to the older one

winder, 2018-05-12:
Did the crash happen right on startup? Sorry about that, I noticed some issues with the way the Visualizer was configured and unfortunately the change has some weird backward compatibility issues.

In my testing the crash didn't happen right away, so closing the extra visualizer was possible. The "Window > Reset Windows" option can also be used to put things back to a good state.

vMeph, 2018-05-12:
the program opens  but it crashes before i was able to close the 2 visualizer
the solution was go to   %appdata% and delete .ugsplatform folder
but now the jog controler window shows has the old one