winder/Universal-G-Code-Sender #1033: Probe

waltoncnc, 2018-05-13:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

closed 2018-05-17 by winder

waltoncnc, 2018-05-18:
It still doesn't work. XYZ works fine but if you do just z it is still off. It goes up to 21.07 mm. That is a little shy of 9/16ths. from workspace. Have deleted all the other builds and installed May 17 nightly build. Also still can't go past 10 in the retract probe settings. Using Windows 7 and Shapeoko 3xxl. Any other suggestions?

winder, 2018-05-18:
Hmm, for some reason it is still loading the old code. I just made another change to see if it can force the new code to load. See if it works with the new update - if all goes well the probe window you have open now should automatically close and you can re-open the new one.

waltoncnc, 2018-05-18:
It appears that probing is working in mm but when you try z in inches it goes down and touches once then goes way up then tries to go down then it runs out of touch limit and throws a error. We prefer to use inches if possible.

jwhelget, 2019-01-20:
waltoncnc I finally found someone with the same situation.   Has this been resolved?   I had to set the probe distance to -1.0" or it would alarm out.  I too want to work in inches.

JacksCrafts, 2019-01-20:
I have the same problem. So I just use the macro instead.
[image]

JacksCrafts, 2019-01-20:
Here is the code for setting Z. I just use an old L bracket I found in a junk box. It is 0.054 inches thick.
Go to options - the UGS tab under there is the Macro tab. fill in the 3 boxes. This code will go down (negative direction) for 1.5 inches before it errorrs out. When it touches the plate it will go up 0.054 plus 0.40 for a total of 0.454 inches above zero. 

Set Z
G20 G38.2 Z-1.50 F2; G92 Z.054; G91 G0 Z.4
Sets Z to zero

jwhelget, 2019-01-21:
Thanks I just posted as an issue.   Would like to use the xyz also but didn't trust that if the z didn't work yet.

JacksCrafts, 2019-01-21:
I am an amateur but this should work.

Corner Probe 0.12 thick on all 3 walls.
G20 G38.2 Z-1.25 F2; G92 Z0.12; G91 G0 Z1.00; G91 G0 X-1.75; G91 G0 Z-1.12; G38.2 X1.50 F2; G92 X-0.12; G91 G0 X-0.88; G91 G0 Z1.12; G91 G0 Y-2.00; G91 G0 X1.50; G91 G0 Z-1.12; G38.2 Y1.50 F2; G92 Y-0.12; G91 G0 Y-0.88; G91 G0 Z1.12; G91 G0 Y2.00

commit 066254d5 (2018-05-17): Fix z-probe and minimum retract distance. Closes #1033, #981, #1005, #893, #878