winder/Universal-G-Code-Sender #1034: UGSPlatform closing itselft after 5 secs.

AbySet, 2018-05-14:
## Problem description 
When i run the application, it's running for 5 seconds then closing itself without any error message.

### Expected Behavior
Keep running.

### Actual Behavior
Closing itself like i a quit click.

### Steps to Reproduce the Problem
  1. Open the program

### Specifications
#### Version
  UGS Platform 2.0 - Nightly Build May 13, 2018

#### Operating system
  Windows 8.1 64 bit

#### Platform
  GRBL 1.1 Flashed into Arduino UNO using Arduino IDE (VARIABLE_SPINDLE in [config.h] file has been disabled to get Z axis switch limit working properly).

#### Other
Everything is working fine with UGS Platform 2.0 - Nightly Build Apr 23, 2018 whitch i still have.

I used a quick screen capture to get this (GCode Platform version does not seems to be updated in [About] box).
[image]

winder, 2018-05-14:
@AbySet I just added a possible fix for this problem. Please download the latest nightly build and let me know if the problem is resolved. Thanks!

winder, 2018-05-14:
You still see the problem with todays build? The problem is due to a corrupt cache, so you wouldn't see it in a clean install. Only after upgrading from an older version (i.e. the April 23 build you have to yesterdays build).

winder, 2018-05-14:
If possible, please don't do the clean install. That will definitely fix it for you, but if you could keep testing I'd like to try fixing it in a way that doesn't require everybody to do that. I'll do some more testing of my own and see if I can find another fix.

When you tested on windows 7, did you run the May 14 build first followed by the Apr 23 build? Or the other way around?

AbySet, 2018-05-15:
Ok i will not do clean install on win8.1, i will keep it and see if your fixes will work.

When i tested on windows 7, i did run Apr 23 first, then i installed May 14 build after and here is the installation history in both PCs:

**Windows 8.1 PC:**
   1. Before May 13 - Running Apr 23 Build, (working OK).
   2. May 13 - **Deleting Apr 23 Build folder** and replacing it by May 13 Build folder  (did not work).
   3. May 14 - **Deleting May 13 Build folder** and replacing it by May 14 Build folder  (did not work).

**Windows 7 PC:**
   1. Before May 13 - Running Apr 23 Build, (working OK).
   2. May 13 - Copy May 14 build folder content into Apr 23 Build folder **replacing previous files** (working OK).
   _May 13 build was not tested in this PC._

hmm seems like there is something in Apr 23 folder that is not replaced on win 7 PC (hidden or teporary file maybe) in copy process and making May 14 build keep running.

Anyway, i will wait for more instructions from you and follow them.

Thank you and see you on the next test.

AbySet, 2018-05-15:
I did not deleted anything and i was running Build May 15 from another location on my desktop (so i dont delete or replace May 14 build files just in case we still need it for testing) and:

YES Build May 15 did work just fine.

I consider then this issue has been fixed in May 15 Build, of course i stay in touch in case you need me to test something else, please let me know if we finished testing so i can permantly remove the previous versions.

If you need me to test incoming versions i will be glad to do it, but please send me a link where to download it cz i'm always confused about where to download latest versions.

Good job and many many thanks for the work you'r doing.

Seya.

AbySet, 2018-05-15:
Thanks man, i'll save the link.

algo1980, 2018-09-22:
i am hit with the same issue and this is kind of very strange one, what's happening at the backend of this issue? please let me know

AbySet, 2018-09-22:
- First get the latest update from here: https://github.com/winder/Universal-G-Code-Sender
- Set your windows folders settings so it shows hidden files.
- Open this folder: C:\users\<Your name user>\AppData\Roaming
- Delete a folder with name: .ugsplatform

This will make you delete all cache memory of the program and clean it all, now you can run it and redo your settings (don'nt forget to set the probe setting if you are using it).

Hope this help.