winder/Universal-G-Code-Sender #1035: UGS Closing after splash screen

poplawa, 2018-05-14:
App closes right after splash screen

I was using UGS platform version for a while and everything was fine. Today my laptop run out of battery and shut down quite rapidly and now I have this problem. 
 - Java reinstalled with fresh version (didn't help)
 - downloaded nightly build 2.0 (same behaviour)

When I run it from cmd, it shows "initializing" and shuts down. 
I'd like to include some other log but don't know how. I marked ENABLE LOGGING in Java configuration panel but I don't think that's it...

#### Operating system
  Windows 7 64 bit
  UGS 1.0.9 and 2.0.0 platform

winder, 2018-05-14:
You see this problem with v1.0.9 and the Platform? Could you run v1.0.9 and from the command line and share the terminal output here?

poplawa, 2018-05-15:
Sorry, my mistake: only "UGS Platform" (the nice interface) but both versions of nightly builds: 
- this one: http://archive.fabacademy.org/2017/fablabbottrophrw/students/66/week12/img/ugs.jpg
- and this one: # cluster incompatible release version21

Classic 1.0.9 works fine, here's its output:

D:\PULPIT\UniversalGcodeSender-v1.0.9\Klasyczny>java -jar -Xmx256m UniversalGcodeSender.jar
maj 15, 2018 8:36:05 AM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Loading settings.
maj 15, 2018 8:36:05 AM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
maj 15, 2018 8:36:05 AM com.willwinder.universalgcodesender.model.GUIBackend add ControllerListener
INFO: Adding controller listener.
maj 15, 2018 8:36:05 AM com.willwinder.universalgcodesender.model.GUIBackend add ControlStateListener
INFO: Adding control state listener.

How can I get a better log for PLATFORM?

winder, 2018-05-15:
This is a duplicate of #1034, I'm working on a fix

winder, 2018-05-15:
@poplawa I have a possible fix available in the current nightly build, please let me know if that resolves the problem for you.

poplawa, 2018-05-15:
Nope... 
The thing is that it just stopped working, I didn't change anything (I was using the same exe file all the time)
My theory is that this sudden reset must have caused some issues on my computer. Either java related or not. I just don't know how to diagnose it. Could you help me with that? 

I saw people posting some detailed logs. I don't know how to enable them...

winder, 2018-05-15:
@poplawa sorry about that, I jumped the gun thinking these were the same problem. I just updated the documentation to describe the log locations, please take a look and share the log file on this ticket. For reference: http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

The log file on Windows 7 is located here:
C:\Users\username\AppData\Roaming\ugsplatform\2.0-SNAPSHOT\var\log\messages.log

poplawa, 2018-05-15:
-------------------------------------------------------------------------------
>Log Session: Tuesday, May 15, 2018 2:50:38 PM CEST
>System Info: 
  Product Version         = Universal Gcode Platform 201609300101
  Operating System        = Windows 7 version 6.1 running on amd64
  Java; VM; Vendor        = 1.8.0_171; Java HotSpot(TM) 64-Bit Server VM 25.171-b11; Oracle Corporation
  Runtime                 = Java(TM) SE Runtime Environment 1.8.0_171-b11
  Java Home               = C:\Program Files\Java\jre1.8.0_171
  System Locale; Encoding = pl_PL (ugsplatform); Cp1250
  Home Directory          = C:\Users\Piotrek
  Current Directory       = C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform
  User Directory          = C:\Users\Piotrek\AppData\Roaming\.ugsplatform\2.0-SNAPSHOT\dev
  Cache Directory         = C:\Users\Piotrek\AppData\Roaming\.ugsplatform\2.0-SNAPSHOT\dev\var\cache
  Installation            = C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\ide
                            C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\java
                            C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\ugsplatform
                            C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\extra
                            C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform
                            C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform
  Boot & Ext. Classpath   = C:\Program Files\Java\jre1.8.0_171\lib\resources.jar;C:\Program Files\Java\jre1.8.0_171\lib\rt.jar;C:\Program Files\Java\jre1.8.0_171\lib\sunrsasign.jar;C:\Program Files\Java\jre1.8.0_171\lib\jsse.jar;C:\Program Files\Java\jre1.8.0_171\lib\jce.jar;C:\Program Files\Java\jre1.8.0_171\lib\charsets.jar;C:\Program Files\Java\jre1.8.0_171\lib\jfr.jar;C:\Program Files\Java\jre1.8.0_171\classes;C:\Program Files\Java\jre1.8.0_171\lib\ext\access-bridge-64.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\cldrdata.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\dnsns.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\jaccess.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\jfxrt.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\localedata.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\nashorn.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\sunec.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\sunjce_provider.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\sunmscapi.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\sunpkcs11.jar;C:\Program Files\Java\jre1.8.0_171\lib\ext\zipfs.jar
  Application Classpath   = C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\boot.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\org-openide-modules.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\org-openide-util-lookup.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\org-openide-util-ui.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\org-openide-util.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\boot_ja.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\boot_pt_BR.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\boot_ru.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\boot_zh_CN.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-modules_ja.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-modules_pt_BR.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-modules_ru.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-modules_zh_CN.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util-lookup_ja.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util-lookup_pt_BR.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util-lookup_ru.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util-lookup_zh_CN.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util-ui_ja.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util-ui_pt_BR.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util-ui_ru.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util-ui_zh_CN.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util_ja.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util_pt_BR.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util_ru.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\lib\locale\org-openide-util_zh_CN.jar
  Startup Classpath       = C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\core\asm-all-5.0.1.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\core\core-base.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\core\core.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\core\org-netbeans-libs-asm.jar;C:\Users\Piotrek\Downloads\ugsplatform (2)\ugsplatform\platform\core\org-openide-filesystems-compat8.jar;C [... cut ...]
	at org.netbeans.core.startup.ModuleList.installNew(ModuleList.java:340)
	at org.netbeans.core.startup.ModuleList.trigger(ModuleList.java:276)
	at org.netbeans.core.startup.ModuleSystem.restore(ModuleSystem.java:301)
	at org.netbeans.core.startup.Main.getModuleSystem(Main.java:181)
	at org.netbeans.core.startup.Main.getModuleSystem(Main.java:150)
	at org.netbeans.core.startup.Main.start(Main.java:307)
	at org.netbeans.core.startup.TopThreadGroup.run(TopThreadGroup.java:123)
	at java.lang.Thread.run(Unknown Source)

winder, 2018-05-15:
@poplawa glad it worked. Using your stack trace I was able to improve the error handling, so if this happens to someone else in the future there will be an actual error message.

closed 2018-05-15 by winder