winder/Universal-G-Code-Sender #1036: Program does IDLE after a long HOLD

poplawa, 2018-05-17:
I paused the program (using physical button connected to Arduino) somewhere in the middle because I had to leave my workshop for about 2 hrs. 
When I came back, the program was no longer in HOLD state but turned into IDLE. The physical RESUME button didn't work. Neither did the jog controls. 
Luckily pressing SEND button resumed the program from current line. 

[image]

I didn't switch off any power supply and my computer didn't go to sleep

winder, 2018-05-17:
What version of GRBL are you using?

poplawa, 2018-05-17:
grbl-1.1f.20170801

poplawa, 2018-05-17:
Windows 7.

winder, 2018-05-17:
It's possible that UGS was somehow confused by the physical pause button being pressed instead of the in-app pause button... but I think a lot of people are using environments like that and I've never heard of this problem.

When using the pause/resume buttons is it possible to pause with the physical button and resume from UGS?

poplawa, 2018-05-18:
Yes. Usually it all works as expected, this was the only time that I saw such behaviour. 
I'll let you know if it happens again

winder, 2018-05-18:
@chamnit have you seen anything like this? I'm pulling the state right out of GRBLs feedback, is there any way it could go from `HOLD` to `IDLE`? I would have thought perhaps something about the USB connection reset, but UGS doesn't have an auto-reconnect and I don't see the version information in the console.