winder/Universal-G-Code-Sender #1037: UGS and Java

mholmberg120, 2018-05-18:
Hi 
I am a total beginner.  I get this error message  when  try to open UGCS.  I have a new computer and I had no problems on my old computer run and cutting with autodesk 360 and UGCS?  I know Java is installed because I can run other Java programs.  please any help would be appreciated? 

[image]

Java Installation not completed
Unable to install Java
There are errors in the following switches:

Java Installation not completed
Unable to install Java
There are errors in the following switches:
C:\Users\mholm\OneDrive\Desktop\MonkeyBirdyDesign\UniversalG
CodeSender.jar
Check that the commands are valid and try again

Thank you
Michael

winder, 2018-05-23:
Does it work if you run using the .bat file?

breiler, 2019-02-20:
With no answer we are not able to help any further, closing this issue.

closed 2019-02-20 by breiler