winder/Universal-G-Code-Sender #1041: Java null pointer exception and no position status updates

ab0032, 2018-05-28:
## Problem description 
Hi,
I get a java null pointer exception and the position is never updated when I manually move X, Y or Z around.
I downloaded and installed the latest stable grbl 1.1 today and downloaded v1.09 latest stable of UGS today. I checked the Arduino pins with an oscilloscope, movement commands are sent.

I also tried
  "export _JAVA_AWT_WM_NONREPARENTING=1"
from the troubleshooting page, but it did not change the behavior.

### Expected Behavior
If I press X+ or X- the position displayed should change. G-code seems to be sent and executed properly.

### Actual Behavior
Position remains at 0,0,0.
On every button press I get a null pointer exception on the console. See below.

### Steps to Reproduce the Problem
Start the program, connect to Arduino and move around in machine control panel.
### Specifications
#### Version
v1.0.9 Nov 15 2015 latest stable

#### Operating system
Debian Linux 64 on intel

#### Platform
grbl 1.1f
java-7-openjdk-amd64/jre

#### Other
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
        at com.willwinder.universalgcodesender.MainWindow.setStatusColorForState(MainWindow.java:2201)
        at com.willwinder.universalgcodesender.MainWindow.statusStringListener(MainWindow.java:2552)
        at com.willwinder.universalgcodesender.AbstractController.dispatchStatusString(AbstractController.java:757)
        at com.willwinder.universalgcodesender.GrblController.handlePositionString(GrblController.java:414)
        at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:126)
        at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:743)
        at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListener(AbstractCommunicator.java:198)
        at com.willwinder.universalgcodesender.AbstractCommunicator$1.run(AbstractCommunicator.java:165)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:312)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:745)
        at java.awt.EventQueue.access$300(EventQueue.java:103)
        at java.awt.EventQueue$3.run(EventQueue.java:706)
        at java.awt.EventQueue$3.run(EventQueue.java:704)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:715)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:218)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:133)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:122)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:118)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:110)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:91)

breiler, 2018-06-12:
Could you try the latest nightly build? You would need to update to Java 8 for this.

breiler, 2019-02-20:
New version of UGS is available and should resolve this issue.

closed 2019-02-20 by breiler