winder/Universal-G-Code-Sender #1043: hardware pendant sending G-Codes to GRBL causing UGS plattform to display error messages

hamster65, 2018-06-01:
Hi,
I am using UGS Plattform Nightly 201609300101 with GRBL. I built a hardware pendant which produces G-Codes for Jogging, Feed Hold, home, probe etc. The pendant's tx wire is connected to GRBL's RX pin with a 1N4148 diode, so both the computer (running UGS) and the pendant can send commands to GRBL (of course not at the same time, it would mess the serial data up).

This works fine with the BCNC software, but I found some issues with UGS in this setup: 

1) When the pendant sends a command, GRBL responds with "OK". This causes UGS to show an error message: "Error while processing response : An unexpected command was completed by the controller."  This has been asked before here: #667. Is there a way to disable UGS reactions to unexpected OKs while keeping all other error messages? Is there a better way to connect my pendant?

2) When I send "!" (feed hold) and "~" (cycle start) afterwards, UGS will show hundreds of error message windows and crash since "~" is not allowed in this state. I think this should be catched by UGS.

Thanks!

breiler, 2018-06-01:
Hi,

1. I recently made some changes so that unexpected commands don't generate this error popup. Instead there's an error message in the console.
2. No idea if this  is resolved as well.

These changes are available in the nightly build, please give it a try and let us know if this works.

hamster65, 2018-06-01:
Hm, that's strange. I downloaded nighty build 29 May but the error messages still popup.
I used the download link "UGS Platform - Universal Gcode Platform, the next generation platform-based interface".

breiler, 2018-06-04:
Ah, sorry. I changed the way it's handling unexpected commands when it's in alarm/error state.
I'm not sure that I dare changing this for ok-responses as well.

hamster65, 2018-06-04:
I guess it is not a good idea to ignore unexpected commands in general, also for example BCNC seems to do that. But maybe as an option?
Another approach for my project would be to create a "serial proxy" which filters out the first ok response after each command my pendant sends.

winder, 2018-06-04:
What version of GRBL are you using? I have noticed that sometimes versions of GRBL 0.9 will send an extra `ok` during startup.

winder, 2018-06-05:
Just took a look and the XLCD profile is no longer enabled by default. You'll need to go into the `firmware_config` directory, copy the `grbl.json` file, and rename `GRBL` to `XLCD` in the new file.

hamster65, 2018-06-05:
I'm using GRBL 1.1 (precompiled release).
I copied ~/.ugs/firmware_config/grbl.json to myXLCD.json and replaced GRBL with XLCD (2 occurrences). XLCD appeared in the drop down menu and it connected, but stills gives the error message.
I just realized that the XLCD project is also using a "man in the middle" approach.

jschoch, 2018-11-11:
I am also messing with a MITM hardware pendant which is having the same problems.  Changing the config to XLCD is not removing the popups.