winder/Universal-G-Code-Sender #1044: 4 Axis Support

devekrehbiel, 2018-06-02:
I tried to leave this on the google forum but it wouldn't let me. This is pretty important! To me anyway. I am making a Robotic Arm that allows for much more versatile control over the grabbing mechanism but requires a 4th Axis. The G-Code is simply to move the arm using 4 different Steppers and one Servo. Your program is perfect for this, of course with the exception I do not see a way to get a 4th axis. Thanks to all who are making the effort on this program.

winder, 2018-06-02:
It isn't currently supported.

Most of the code uses a `Position` object which could be extended, and new axis features are using an `Axis` object. Hopefully, these should be extensible some day.

devekrehbiel, 2018-06-02:
I know and I am sure you have better things to do, but imagine the ability to duplicate any number of "axis" you wish. Like, x1, x2, etc.. just by duplicating the programming code with an extender. This would allow your program to be used in very large manufacturing applications where robotic arms with many stepper motors (or whatever they use for motors) are used. I do not expect you to reinvent the wheel or make it harder than it needs to be.. but if all this is about is motor control via Gcode, you are way more than halfway there. Thanks Winder for your hard work. It's people with the knowledge to get this far that get picked on for MORE! Take care and thanks for listening!

justinclift, 2018-07-06:
As a thought, would this 4-Axis support be related to #1068?

winder, 2018-07-16:
Working on this in #1068

closed 2018-07-16 by winder