winder/Universal-G-Code-Sender #1046: Serial port not visible on macOS High Sierra

classbproject, 2018-06-04:
I have a grbl board that comes with a Silicon Labs CP2104 USB to UART chip. I have downloaded the drivers for this from [here](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers). When I open any version of UGS, it says 'no serial port found'. However, through Terminal and through the Arduino IDE I can see that the port is available. In fact, I can even download sketches through the Arduino IDE. Why is this happening and how do I resolve this?

breiler, 2018-06-04:
In the nightly build we have added a new connection driver that you could try. See the instructions in the issue #940

Let us know if this works for you

classbproject, 2018-06-04:
This works. Thank you.

closed 2018-06-04 by classbproject