winder/Universal-G-Code-Sender #1047: UGS Platform not opening

wolfspaw, 2018-06-05:
When I open UGS Platform it loads half way then closes

winder, 2018-06-06:
Please provide the log file located here:
http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

breiler, 2018-06-18:
Interesting, could you post the contents of the file:
C:\Users\adam\.ugs\UniversalGcodeSender.json

The work around would probably be to delete this file and restart. But it would be nice to know how this happened in the first case.

wolfspaw, 2018-06-20:
I can’t find that file. I’ve also deleted everything and tried to download it again. Am I missing something that I need to delete?

wolfspaw, 2018-06-20:
I’ve deleted everything again and downloaded ugs platform again and everything is working now. Thanks for the help.

carneeki, 2018-06-21:
@winder - I've noticed a couple of tickets that were solved with deleting
the config files (perhaps key/value pairs change over time?). I wonder if
it's worth putting this hint in a troubleshooting guide.

winder, 2018-06-21:
@carneeki I think the latest version should detect the error and re-initialize everything to the default values. I'm not sure why this same problem cropped up a couple times recently though.

closed 2018-06-21 by winder