winder/Universal-G-Code-Sender #1048: Feature request: Pre-start checklist

raechten, 2018-06-06:
## Feature request

In the Inventables Easel software, there is a sort of wizard when you want to carve. It guides you through several steps before starting the actual work (Confirm the thickness of the material, Is the material secure, Confirm the bit to use, Is the spindle on (if you manually turn on/off the spindle), etc.). Although some of them will get annoying, they are helpful.

I'm switching over to UGS Platform and I found that especially the one asking me to make sure the spindle is on is useful to me, as I already had 2 occasions where I forgot to turn the spindle on and had it plunge into the material with the bit not spinning (no damage done, fortunately).

I would like to have a checklist popup when clicking the Start button. This list would be user editable (in settings) and would have a checkbox after each item. Once all checkboxes are checked, the job actually starts.

This checklist only pops up if you have enable this in the settings (or optionally, only if you have items in the checklist in the settings).

khturnings, 2018-09-04:
Would like to add to these features - 
1. Once the job is zero'd it checks a loaded gcode file does not exceed the min/max size of the machine (or in the bounds of machine work area)
2. I had some jobs miss a step. As I stopped the job this did not cut into wanted material but I would have liked to be able to skip to a point in the gcode just before the dropped step and then resume the job instead of re-running the complete job.

BallscrewBob, 2018-10-03:
+1 for khturnings suggestion similar to LaserGRBL options.