winder/Universal-G-Code-Sender #1049: Add usage statistics

breiler, 2018-06-09:
## Feature request
I would like to start collecting application usage statistics from the application so that we can learn what functions are being used and how. I think that by collecting this  data will help us make more strategic decisions on what functions that need attention, get insights on what should be improved or ideas for new features.

 - [ ] The statistics should not contain any personal information or any data that could identify a person. 
 - [ ] It should be opt-in only, the users must be allowed to decide if they want to contribute with their data
 - [ ] Reports and statistics should be publicly available

Example of data to be collected:
 - Application open / close
 - Job start / pause / stop
 - Job duration and number of rows
 - Usage of actions such as Homing, Return to zero, Check mode, Reset alarm...

BallscrewBob, 2018-06-13:
Auto reporting of failures  i think might be worthy of inclusion but those stats should probably not be public as such...REALLY tells you where to aim if the same thing is a common denominator

breiler, 2018-06-16:
Added a popup that will state a question the first time UGS is started:
[image]

It's a bit annoying to get a lot of popups so I added a checkbox to disable the nightly build warning:
[image]

It's possible to enable the usage statistics in the sender settings:
[image]

justinclift, 2018-07-11:
> I would like to start collecting application usage statistics from the application so that we can learn what functions are being used and how.

As a general thought, be **very** careful of this.  One thing that seems to commonly happen after this gets done, is that applications start to lose their advanced/power-user features.

My best guess is that non-newbie users generally turn off the application statistics, so the "what gets used most often" stats become badly misrepresented.  Thus triggering the removal of things only used by said advanced/power-users. :frowning:

Not sure if there's a way to correct for the misrepresentation of stats... maybe applying some kind of normalisation function to them before analysis?

Automated crash reports though, as mentioned above by @BallscrewBob seem like they're be universally useful. :smile:

breiler, 2019-06-12:
It would have been sweet to have some usage statistics. But the privacy concerns maybe outweighs this... Closing this.

closed 2019-06-12 by breiler

justinclift, 2019-06-12:
What level of usage stats were you hoping for?

As a general thought, if you have an automatic check for new releases (which can be disabled of course), that can help give a rough lower estimate for the number of daily users.

We (https://github.com/sqlitebrowser/sqlitebrowser/) already had a version check happening, so turning that into interesting stats wasn't too hard: https://sqlitebrowser.org/stats/

breiler, 2019-06-12:
We made a PoC where we pushed a bunch of data to our own instance of Matomo.

I wanted to know:
* Edition of UGS (Platform, Classic, Other?)
* Version
* Number of unique users
* Optionally which functions being used

So we basically got this up and running but never merged it.

justinclift, 2019-06-12:
Yeah, the functions being used bit would definitely need user agreement.